Geektoolkit/Dynaframe3 #1: Initial feedback

DFF-fred, 2020-09-01:
Hi!
Thank you for the follow up and the nice upgrade! I've ran this under a RPI3 as this is the hardware I'll be using for an upcoming project. Works like a charm (compared to v1 on rpi4...). 
Now I have a some feedback based on what I could get done, or not:
- Writing the setting URL on start is a security issue as any "viewer" could then go and mess up things through that 'unprotected' web interface. There should probably be an option to have it, or password protect it
- Could figure out how to change the default Pictures folder to something else. I clicked on the "All in: /home/pi/Pictures" but nothing happened except a refresh (using FF).
- Uploaded a mp4 video through scp to Videos, restarted, nothing appeared. Copied the video to Pictures, video also didn't show after a reboot. Tried the submit a few times but not sure if this is refreshing the file list or not.
- The screen went black after a while, rebooting didn't change that (ssh still worked). I unplugged and replugged the monitor and that fixed the problem. I'll need to test longer to see if the problem is on my side or somewhere else.

I didn't try allwaysync yet as I wanted to see basic functionalities first. Currently my main issues are playing videos and switching active playing directory from the web interface.

Thank you again for the work you did!

Fred
ps: Dynaframe is listed at ~250% CPU utilisation under top but the whole thing seems very responsive.

Geektoolkit, 2020-09-02:
Whoa, you're running way ahead of me! I JUST got this up and working myself :)
Thankyou this is great feedback, and likely work that just needs to be done. but hopefully it's miles ahead of the last release :)

1) It' uses 'avalonia' which is a pretty heavy framework, but lets me work in a MUCH cleaner way (compiled classes, cross platform, c#, etc) This is probably why you're seeing the 250% CPU usage. Believe me though, the speed at which I can add features now, it'll be worth it :)  

2) For the startup listener..you're absolutely right. I was thinking of a way to try to make it user friendly, especially since reboots could change the IP.  I'll think on that, perhaps there needs to be a password on the webUI no mater what. I figure since there isn't necessarily a keyboard/mouse on the device, I wanted a way for an admin to know what the login url was.  I'll think on it. Password protecting the page may be the best option to balance security and usability.

3)The videos folder isn't used currently.  I hadn't thought of that. I think what i need is an aggregate of multiple directories for it to look at, and then a way to add more. I'll get that added in, it just makes sense.

4) The feature of adding a directory isn't in yet.  I have to be careful there...if say I add a usb drive (one of my ideal scenarios) and then it gets removed, then I have to figure out how to gracefully remove images from the playlist. I have ideas on this, but want to take time on how to properly implement them.  I'd also like to support NAS drives, UNC shares, etc. Allsync may not work on this version since I don't have samba support (Unless you set that up yourself).  Rsync may be interesting for you to try...you could run it on the pi and have it sync there without another machine. I love that sceanario for that "I got one for my parents, and they can see their grandchilds baby pictures magically' scenario.

5) Screen turning off - that's a bug, I found it last night and think I have a fix. I'll try to get an update out in a few days with a fix for that one.

I'm going to keep this open so that i have a checklist to work through, all of these are things I'd like to fix or investigate further.  Thankyou for the feedback!

DFF-fred, 2020-09-02:
Wow great! Thank you for the feedback on the feedback ;-)
Small question related to Rsync, UNC shares and all: do the images/videos need to be on the SD card or could they be loaded from a network share, without writing anything on the SD card? I mean it would have to be somewhere in a cache/temp folder, especially with larger vids, but simply getting the files off the network maybe after a start/submit/refresh sequence without reading anything from the SD card?

Thank you again.

Fred

Geektoolkit, 2020-09-02:
hmm right now they'd have to load off the SD card...though you should be able to usb boot a pi3B+ (and there's ways to do it with a pi4 as well).  USB Drives I hear are more reliable than SD cards typically. As for loading it remotely that's not implemented yet, but I'll look into it when I get there. I think to support NAS drives I'd have to load from the remote source.

qwksilver, 2020-09-09:
#5 I was connected via vnc when the screen turned off, moving the mouse on the vnc appears to rewake the screen.
i'm going to look for a way to turn off the screen saver.  i'm opening another issue i'm having at this time also

Geektoolkit, 2020-09-11:
Ok I have an 'alpha' release with a bunch of work in it. If you get a chance please try it.  It includes:

1) A command to keep the screen turning off. Should solve '5' above if it worked.
2) There is an 'add folder' section now...you can use that to add multiple folders (videos, a usb drive, a nas drive, etc)

Note: You can't remove a folder yet, and you can't remove the default folder (my pictures) yet.  That should be in the next drop though now that i have an engine that can handle multiple folders.

The security issue of the URL is one I'm working on tackling, as it's a bit tricky. I think I have to password protect the website, and since it's not a 'real' webserver that is going to take some thinking. I'll close this once it's tested and confirmed that the rest works, and open a seperate issue for just the security portion.

qwksilver, 2020-09-30:
5: I'll note that i've had some fairly large runtimes and no screen loss, not sure if the code update 2.04 helped, or if turning off "screen blanking" in raspi-config was the trick

Geektoolkit, 2020-10-01:
I think turning off screen blanking is doing it. I am going to make that just a thing that is part of the setup, it's not terribly hard, and there's a command line already that has to be run.  It'll be much more reliable than me messing with settings.  

The only feedback item left on this is turning off the ip address on start. I'm planning that for next release as I've gotten two requests for it, and I get why.

qwksilver, 2020-10-01:
actually i think part 1 is Writing the setting URL on start is a security issue as any "viewer" could then go and mess up things through that 'unprotected' web interface. There should probably be an option to have it, or password protect it

i created a collected settings wishlist that moves the remaining issue here to there, i think all of this is done, and can be closed.

Geektoolkit, 2020-10-26:
Closing this out. 2.06 has the option to turn off the url at start by setting the time to '0'.  2.07 should improve stability and complete this task.

Note: Screen blanking is going to be an install step...sudo raspi-config, under advanced there is an option to turn it off. The ability to turn it on/off dynamically is an open issue being tracked thanks to qwksilvers work

closed 2020-10-26 by Geektoolkit