Geektoolkit/Dynaframe3 #2: Powering the Pi

DFF-fred, 2020-09-01:
Hi!
This is a bit OT but I've read conflicting information between various online communities. Trying to avoid a lot of cabling I'm thinking to power the Pi from the monitor USB port(s) to reach 1A or maybe more. 
Then I also need to have some kind of small UPS connected (as we have regular power cuts) which could send a signal to the Pi (similar to what NUT does) and tell it to switch off. Do you have any experience or recommendation in the matter?
Thank you.

Geektoolkit, 2020-09-02:
The pi3 and 3B need 2.4 amps minimum...USB ports spec at 1amp, so it's rare to find a monitor that will be able to properly power a pi directly.   If a monitor has 2 usb ports you could rig up a cable from both, combine them, and get 2Amps potentially, but you start getting risky.  One other option would be to crack the monitor case and put it into a wooden frame...changing up the look and giving you space to work, then using a buck converter to try to get an internal voltage (12-24 likely) down to 5v.  I use wire covers anyway, so for me running the pi cable seperately is actually just as clean, and more convenient.  

For instance, my magic mirror uses a motion sensor to detect when to turn the monitor on/off..but to do that the pi has to stay powered.  The pi power draw is maybe $10USD a year so I don't mind leaving them on, but I do like that they can turn the monitor off (Not a feature in dynaframe yet, but it'll get there).  To do that though, I need seperate power for the pi and monitor.

I've heard of UPS solutions for the pi, but haven't researched them. I think instructables had a DIY solution depending on how deep you want to get into it.  I'm not familiar enough with that part to really comment sadly.

DFF-fred, 2020-09-02:
Thank you for the hints, I'll definitely look into Instructables. Opening the monitor is not an issue and therefore probably better if I go that way. My main concern is really the Pi not properly switching off and having the SD card unusable the next day. 
Thank you.

Geektoolkit, 2020-09-02:
This is the one I was thinking of, but I've not tried it. It may give an idea of a way to get things going. This one looks like there's a link at the end that got cut off sadly. https://tutorials-raspberrypi.com/raspberry-pi-build-your-own-ups-uninterruptible-power-supply/

Geektoolkit, 2020-09-04:
Going to close this one as I'm about to release on youtube, and want to keep my issue list clean as I expect I'll get flooded.

closed 2020-09-04 by Geektoolkit