Geektoolkit/Dynaframe3 #3: Laptop install

n4tur3502, 2020-09-01:
Hello, I am trying to get this to run on a laptop running the raspberry pi os. It keeps giving an error on the log "/lib/ld-linux-armhf.so.3: No such file or directory" Is there a better way to get this to work on a windows laptop?

Geektoolkit, 2020-09-02:
Are you compiling the code and running it or are you trying to use the install script? If you're compiling it and running it should 'just work' on Windows.  If you're downloading the release, it's for Arm32bit linux only right now (I'll have a windows release up hopefully in a day or so, working on a youtube video right now). Let me know, but we should be able to get it running for you

n4tur3502, 2020-09-02:
Omg, thank you so much! I love this program an inordinate amount lol.

Geektoolkit, 2020-09-04:
2.02, a windows 10 release, has been published! Please give it a try and let me know if that fixes your issues. Unzip it, and then make sure you have photos in your 'my pictures' folder, launch the .exe and you should be up and running.

closed 2020-09-04 by Geektoolkit