Geektoolkit/Dynaframe3 #4: V203 Issue list

Geektoolkit, 2020-09-08:
I'm opening an issue to track the items I want to tackle for V203:
1) Add the ability to add arbitrary paths to gather photos from. This should enable it to use any mnt\ on Linux, or directory on windows.  This should also allow for the ability to support NAS drives, USB keys, etc. provided the user sets them up
2) Investigate sleep and ensure that that bug is squashed.  Possibly edit run.sh to ensure that sleep doesnt happen
3) Add keyboard control for things such as showing exif info, fast forward/reverse (Prepping for 204 release with gesture control using CQRobot or IR control)
4) General bugfixes as they appear

NOTE: 202 has the initial bugfix for sleeping/monitor off, as well as scaling of videos correctly

Geektoolkit, 2020-09-10:
Update: 202 didn't fix sleep issue, going to have to update run.sh to turn off sleep for the session.

qwksilver, 2020-09-10:
i turned off screen blanking and sleep doesn't occur on my setup so far, i'm uptime about 30 min, so maybe script in to disable screen blanking in pos (maybe i should us pios as the acronym...hmmmm....)

Geektoolkit, 2020-09-15:
LOL! What script did you use? I tired one using autostart and set:
@xset s off
@xset -dpms
@xset s noblank
which was terrible..it crashed the shell, broke Dynaframe and a few other bad things. I didn't have xscreensaver installed, so maybe that was it. But if you have a better script please let me know and I'll try that :) 2.04 sure as heck hasn't fixed this either.

qwksilver, 2020-09-16:
no script, i just went into sudo raspi-config i think and turned off screen blanking manually.

Geektoolkit, 2020-09-30:
I'm just going to add 'use raspi-config to turn off sleep' to the instructions. I keep stumbling trying to do it and just hitting new issues.  Going to close this also as everythign is addressed except the keyboard issues which will be a new feature in the future.

closed 2020-09-30 by Geektoolkit