Geektoolkit/Dynaframe3 #5: Video, filetype & resolution

mcfrojd, 2020-09-08:
What kind of video files works in Dynaframe?

I was looking to use dynaframe as a fake fireplace and tried a cinemagraph fire mp4 file from https://www.pexels.com/video/burning-firewood-to-ashes-5197737/
but i cant see it start on my pi3 and vnc?
.jpg files works fine.
And the dynaframe log says:
```
setterm: $TERM is not defined.
DF Playing: /home/pi/Pictures/519773.mp4
have a nice day ;)
```
but the clip dont seems to start.

Geektoolkit, 2020-09-08:
It uses omxplayer to play video files, so anything that omxplayer plays should work. 

I tried the default download (4k) size and omxplayer showed essentially what you saw...I got the 'have a nice day' which comes from omxplayer.  

I then tried the 1080p size and it worked for me.  Can you double check the size you're running, and try it in omxplayer.  If you're vnc'd in, remember it wont show on the vnc screen only the actual frame (has to do with the way linux handles terminals). 

Hoping the 1080p thing is the fix, if not let me know.

mcfrojd, 2020-09-09:
Yes the 1080p version worked :-)
Is there any way to make the restart of that clip (im running only that 28 sec file) to have a smother restart?
Now every 28 sec i get a black raspberry pi desktop, the dynaframe clock and the mouse pointer to show  for a little while before the video starts over.
Tried to set transition to "0" and turn the clock off but no diffrence.
(the clock is not visible when the video is running) 

Maybe a continuous loop is imposible to run?
Im downloading a longer file that im going to test, but im not sure the pi will handle a much bigger file.
While running this 28 sec file my pi runs at 78 degrees celsius (172.4 f) and the cpu use is quite high.

mcfrojd, 2020-09-09:
I had a look at [video_looper](https://github.com/adafruit/pi_video_looper) and when using that and changed from their default player (omxplayer) to their "hello_video" player i get an acceptable loop from that file.
And the pi runs at 56 degrees celsius (133 f).
So that solution will be better for my fake fireplace. But i have some upcoming digital art projects that will use your Dynaframe3.
Thanks for your feedback, until next time.

Geektoolkit, 2020-09-11:
mcfrojd - Thankyou for pointing that out. I'm working with qwksilver's feedback on the video playback time as well.  It seems that possibly looping the video until the slideshow timer has expired may be the desirable way to have this operate.   I'll work on that so that it's an option in the future.

Geektoolkit, 2020-10-01:
Closing this. If someone wants to loop videos, Video Looper is the way to go. In the meantime i'll make options for 'play till end' or 'play for duration' to the video engine, and then if the duration > Video length I'll loop it.

closed 2020-10-01 by Geektoolkit