Geektoolkit/Dynaframe3 #6: Documentation: Exit/Close

qwksilver, 2020-09-09:
I found a way to exit to desktop (ESC) to adjust rclone, and screen saver settings, however as a user of pi/linix with almost no experience, I tried Ctrl-C, Ctr-X, and a few other combos a lot, and only thought of esc after about 1/2 hour.  total face palm moment, but i was reading through the documentation and no mention of how to stop was shown.

as an aside to that, restarting without rebooting

Geektoolkit, 2020-09-10:
Thanks this is good feedback, I've updated the wiki and the readme with this info.

To restart, simply launch Dynaframe form the pi\Dynaframe folder.  I'll work to make it easier to figure out.

closed 2020-09-10 by Geektoolkit

qwksilver, 2020-09-11:
suggest add launch shortcut to taskbar and menu