Geektoolkit/Dynaframe3 #7: Impressions & Requests

qwksilver, 2020-09-09:
1: Would like to define settings per folder, even in random, specifically rotation is sorta folder specific for me.
2: Scaling Not always desired, maybe add to per folder settings. (note below)
3: multiple layers of folders  Landscape set, and portrait set, with subs by name.  currently only one layer  and "all" are activateable.
4: it might be nice to change the clock location on a per photo basis somehow (off, ul, ur, ll, lr) type menu
5: changing the font and color of the clock would be nice too
6: a gif file in one folder is constantly skipped, so don't know if bugged or not doable.

the way pi3d was handling scaling photos was to scale to the narrowest dimension then create a fade/fog effect from the photo to fill the missing part of the screen. something like this photo:  https://i.stack.imgur.com/I5xoL.jpg  i would like that option for when landscaped photos show on a portrait setup and vice vs.

Geektoolkit, 2020-09-10:
Thankyou this is a good list.  Per folder options are interesting.  I think the fade/fog/blur effect should be something to get working sooner than later..that'll definitely enable some of the other options here. I'll look into how to do that to unlock these other features.

Archonw, 2020-09-27:
Maybe buttons for reboot and simple display off and on would nice at the web ui.

qwksilver, 2020-10-28:
all issues here are either resolved or listed/discussed elsewhere  (2.07)

closed 2020-10-28 by qwksilver