Geektoolkit/Dynaframe3 #9: 1 Bug (two videos), 1 issue (lost config)

qwksilver, 2020-09-15:
version 2.04 preview
issue first
each reinstall, or new version install, requires complete setup of the options screen, no biggie now but as options grow it will become an issue.  recomend a restore previous config, manual backup, restore manual backup, and a reset to defaults, possibly a dedicated settings screen

new ui big improvement.

video issue was an accident, i noticed that my vnc isn't showing video so i can see the slide changes (in the background) there. I also accidentally set the slide shift to 700 ms.  so the slides are ripping past, while the video is playing on the main screen, and then.  another video tries to run concurrently, it causes a mad epileptic fit inducing flicker of the screen, but thankfully not a crash.

recommend a blank background slide be loaded when a video is loaded, and then a kill command ( sudo pkill omxplayer) when the script opens the next slide. this will protect the multivideo issue which could still occur in a small director with longer video scripts. (even without the boo boo of a 700ms slide time)

Geektoolkit, 2020-09-15:
I'm seeing the epilieptic behavior, thankyou I think finally it makes sense why that's happening! I'm going to rewrite the vide/movie engine next so should get that fixed up.

My next update will include 'remove' custom folders..the 2.04 you have should allow you to do stuff like add a custom path and then use a usb drive for content.  Glad you like the UI, I'm terrible at design but had to do something to what I had. Ideally it'll have thumbnails for each folder at some point.

I'm wondering if I should limit the settings to just seconds...so then there's no confusion about 700 vs. 7000 for intervals, I could lock everything to '7 seconds' or '70 seconds' and then it's super clear how long stuff is set for.  Let me know your thoughts.

qwksilver, 2020-09-16:
I think seconds would be fine on slide change, but my sweet spot on fade is the 1600 default.  pi-4.  I think there is a way to make sure omxplayer only has one instance in the scripting.  There was some code [here](https://github.com/misperry/Cloud_Sync_Picture_Frame_Raspi/blob/master/Scripts/Get_Pictures.sh#L2) that exited if a script were already running.  this might be adaptable to that purpose, only in this case send the pkill before launching the new video slide.

optionally just pkill omxplayer before each slide change would enforce the slide duration uniformly.

Geektoolkit, 2020-09-23:
Makes sense. I'm finishing up another video/project then will dive back into this, and i'm going to rewrite the video engine with this feedback.  I'll try to go to seconds but see if I can support decimals, so that people can use 1.6  or .5 to be fancy, but for others it'll just be a simple whole number.

qwksilver, 2020-09-30:
honestly i'm not at 14 and 2 so..... full seconds might not be that terrible.

Geektoolkit, 2020-10-26:
Please open a new issue for the settings migration on updates. I do want to do that one, it's important, but I want to make sure i get it right. 

I've rewritten the video engine. I don't do the black background yet, that'll be in the next build, but please try 2.07 and we should see alot less video crazyness. I kill the video player now after the slide timer expires.

qwksilver, 2020-10-28:
leftover issues moved to 2.07 compilation, new video engine now in debug phase, new tickets for new problems.

closed 2020-10-28 by qwksilver