Geektoolkit/Dynaframe3 #10: appsettings.json Foo Bar

Torxsmind, 2020-09-21:
Default download has "Foo Bar" listed.  Dynaframe launches with a big error instructing to check appsettings.json

Geektoolkit, 2020-09-23:
Thankyou, yes. I've been using the website to save/update the profile and missed that it's starting out on fresh installs as invalid. I'll get a valid one in there.

qwksilver, 2020-10-01:
i had this happen after I deleted the /home/Pi/Pictures, now I find that rebooting fixed the error, but also puts  /home/Pi/Pictures back in my config screen.

Geektoolkit, 2020-10-01:
I'm going to update the default to have /home/pi/pictures instead of the foo/bar and then I'll test to make sure that sticks. I'll also add in other defaults. The foo/bar was a sample thing I didn't think would affect anything as it's designed to get overwritten by current settings immediately.  Thats what I get for thinking :)

Geektoolkit, 2020-10-26:
Ok, did some testing, this shouldn't affect anything with 2.07. The appsetting 'file' is still foo/bar but in memory it gets filled with options that get used. If you make any changes they get written over the appsetting file and all should be well. I now log things to the console so if we hit issues please let me know. Resolving this as 'no longer repros' in 2.07.

closed 2020-10-26 by Geektoolkit

NbrTDB, 2022-04-28:
It would be handy if the appsettings was preloaded with sensible defaults, so we can easily see what needs to be in there to make any changes manually. For us users affected by #117 it's always foo/bar even when we try to change settings in the web UI, and settings are not persisted across reboots (I'm guessing they're only being held in memory, based on your above post).