Geektoolkit/Dynaframe3 #13: Complicated Idea: Rotation Detection:  lots of ambition here

qwksilver, 2020-09-29:
**reasoning, let's make a desktop picture frame, now we set it down and it's landscape, but then we pick it up, and set it back down portrait.  let's automatically detect the change and use appropriate photos, in the correct orientation.  second reason, manual rotating wall frame, third reason automatic rotating wall frame.**

possible electronics: https://www.pololu.com/product/2129, or similar

phase 1: create a selector for portrait, landscape, and to set the appropriate screen orientation for each (0,90,180,270)

phase 2a: separate configs space for portrait and landscape (including relative orientation), specifically what directory to use for landscapes and what to use for portraits (assuming all subdirectories too), and then allowing existing functionality to use less photos by clicking a sub directory, and only showing the directory tree valid for the running orientation

or (and i like this better)

alt phase 2b:program randomizer/player to index height width and compute ratio and only include tall photos in portrait and wide photos in landscape, also allowing a variance ratio and 1:1 up to a programable 4:3/3:4 in both modes.  

phase 3: allow command line parameters to start dynaframe in a specific orientation (allows another application to kill and restart to apply orientation change externally)

phase 4: detect orientation from sensor directy and switch running mode on the fly.

phase 5: add rotation schedule to config, and send command to have servo change orientation of auto frame, add button to webui to do same, change mode and apply rotation servo automatically.....

phase 6: upgrade servo's and sensors software, and rotation steps to create perfect level/plum on each auto rotate

phase 7:  45degre, 30deg, 60deg rotations and photos  :> because art! (would require custom directors and configs from 2a)

I release all these ideas to this project unconditionally.

Geektoolkit, 2020-09-30:
I've thought a bit about rotation and supporting it. There is a 50 dollar mount on amazon that supports rotation, and I thought if I did my own 3D print design I might be able to bring the price down to pennies.  Ironically the code for this is relatively simple since I already have a 'runtime rotation' concept in it.  I'll think on it, but this is going to be lower pri than things like the movie engine and your playlist idea (which is killer by the way, I've been racking my brain of the best way to implement that).

qwksilver, 2020-09-30:
I'd be interested to see the amazon option.

Geektoolkit, 2020-10-01:
here ya go, had to dig it up: https://www.amazon.com/Mount-Landscape-Portrait-Low-Profile-Westinghouse-x/dp/B005MHGCSK/
I used it in this video: https://www.youtube.com/watch?v=YJWmkb-lacA

I've seen servos do monitor rotations for arcade monitors, so a flatscreen seems like it'd be easy/doable. 

Still need a sensor like you said.  What I'm going to do is keep it in mind for now.

qwksilver, 2021-02-06:
This item found recently that may make manual screen rotation detection and adaptation possible, if someone is looking to do an art display, or pehaps a desk frame that you can orient two ways, and might want to utilize this function then comment here otherwise it will likely stay a really long term goal.

item for memory [Youtube , esp32 mpu-6050 accelerometer gyroscope](https://youtube.com/watch?v=dXcF-Uqa-gw&feature=share)