Geektoolkit/Dynaframe3 #16: Collected Requests: Settings Revamp

qwksilver, 2020-10-01:
Existing
1: Separate settings screen
2: option to hide ip during startup
3: a more secure way to manage and collect settings
4: settings specific to each directory

New
5: Web Pause button that can instantly freeze frame the live picture until resume pressed (add to smarthome api later)
6: reboot now button to go with shutdown buttons

mulscully, 2020-10-06:
In looking at 101 above.  I would like to see physical button on the frame for pausing.  It's going to be my wife, who sees an image and want it paused for a while and she is not going to open the web interface to do it (Although that option is nice too).  I am thinking a pause/resume button hooked to a GPIO pin,  Maybe have the user configure which pin in the settings? or maybe just pick a pin.  Maybe even a Prev and Next button.

Also,  would be nice to set a time to turn on TV and turn off TV.  Say 3:00Pm turn on when coming home, 11:30 PM turn off TV after heading to bed.  Ir better yet, motion sensing to turn on/off or standby TV.

Geektoolkit, 2020-10-26:
I'll look into the on/off times, however this may just be something a smarthome implemetnation is better for. I used to use a smart outlet to turn it on/off and that worked well.

for 101, I have the beginning workings of this in there in 2.07, but it's not complete yet. If you want a physical button I suggest something like a number pad style keyboard...I can then map buttons to that so you can control it with something simple.  Unless you mean like a GPIO wired to the pi..I can look into that. The one thing I've added in 2.07 is you can use arrow keys to go forward/back which is nice.

for 3 I have the option to hide the IP, beyond that I"ll look into how to do a password setup. I don't have a true web server so it's very hacky messing with the webUI. 

4 I'm going to punt on. I did some test UI layouts and I don't see a way to do this that doesn't wreck usability, sorry.  I may revisit, but unless I get alot more users needing per folder settings, it's not going to be something that I want to take on.  

6 and 7 I'm looking into. I don't run as sudo on the pi so I may not have the ability to actually control that.

mulscully, 2020-10-27:
Yeah,  I was thinking maybe something where we can take advantage of the GPIO and map some functions to pins

qwksilver, 2020-11-22:
relabeled 1-7 101-103 to a-j for consistency in ferences from other issues.
#16g implemented in 2.09 works

Geektoolkit, 2020-11-25:
Screen on and off have been added as of 2.10.  2.11 adds pause as well via keyboard 'p'.  A GPIO to key press mapper in python should allow this to be done now. It'd be a project, but anything involving the GPIOs will be.  closing this as completed with the understanding that you'll have to write a snippet of python to use it.
https://nitratine.net/blog/post/simulate-keypresses-in-python/ can show you how to inject the keypresses via pyinput.  From there it's listening to the GPIOs and mapping them to the keypresses. This can currently give you pause, Forward, backward, and 'go to first' image.  Also it'll give you the ability to turn on the file info bar, set it to 'date/time' mode, or to hide it.  As I add keys more functions can become available, but I figure this is a start. And get's you the main one you asked for..pause, and file info :)

closed 2020-11-25 by Geektoolkit