Geektoolkit/Dynaframe3 #17: Feature Request: Better folder view.

qwksilver, 2020-10-02:
I have two folders coming over from dropbox
Portrait (with subs Anime, B, K, Web)
Landscape (with subs (Wallpaper, Down, Sales)

here is what my gui looks like on every reboot
[image]

i get home pi pictures back with each reboot but i only want the two subs as feeds
[image]

so issues requests here:
1: want to delete default dir permanatly
2: want uniform box sizes for each box/column
3: this is very jumbled i can't tell what directories go with what folders and so on. so request to show in more like this:
[image]

this is just an image edit for the concept, but the goals here are 

- another, lighter, darker, or different color for each "root" 
- and a offset for the sub directories line/lines
- a clear space between each "root"
- defined min/max button sizes
- a section, likely better moved to settings screen, (#16) to remove/add new directories

qwksilver, 2020-10-02:
oh and i was hoping @tmckay1 could contribute some ideas on how to achieve this.

tmckay1, 2020-10-02:
Sure I am a bit busy this week but will try to come up with some suggestions by the weekend. Would be nice to clean up the GUI for the directories

tmckay1, 2020-10-04:
Shouldn't be that hard to do. Not sure if you would want to not show the default though since it is clearer to the user what is going on. If it doesn't show, then someone might not think we're looking in that directory, but you could display some hint text underneath saying we are searching the default.

As for trying to display it like you have in the image, it would be very easy to do that. You would just have to return a hash here: https://github.com/Geektoolkit/Dynaframe3/blob/master/SimpleHTTPServer.cs#L392 instead of an array of directories. The hash would be the directory path pointing to an array of directories that are subdirectories, so like:
```
{
  "/Users/pi/Pictures" => ["Landscape", "Portrait"],
  "/Users/pi/Pictures/Portrait" => ["Anime", "B", "K", "Web"],
  ...
}
```

Then when you make the links you just need to encapsulate those into rows for each entry and appropriate classes for the keys and values. To be specific, once you have the hash above, it would be easy to traverse that structure to create an html string like this:
```
<div class="directory-row">
  <div class="directory-parent"><a href=?dir=BLAH class="directory-button-sub">Pictures</a></div>
  <div class="directory-subs">
    <div class="directory-sub"><a href=?dir=BLAH class="directory-button-sub">Portrait</a></div>
    <div class="directory-sub"><a href=?dir=BLAH class="directory-button-sub">Landscape</a></div>
  </div>
</div>
<div class="directory-row">
  <div class="directory-parent"><a href=?dir=BLAH class="directory-button-sub">Portrait</a></div>
  <div class="directory-subs">
    <div class="directory-sub"><a href=?dir=BLAH class="directory-button-sub">Anime</a></div>
    <div class="directory-sub"><a href=?dir=BLAH class="directory-button-sub">B</a></div>
    <div class="directory-sub"><a href=?dir=BLAH class="directory-button-sub">K</a></div>
    <div class="directory-sub"><a href=?dir=BLAH class="directory-button-sub">Web</a></div>
  </div>
</div>
...
```
once we have that it would be super easy to use CSS to format the structure into something much nicer. I would also encapsulate all of this in its own class.

If you didn't want Pictures there you shouldn't include it in the hash originally. Not sure if that makes sense though since we are technically searching that directory for photos.

qwksilver, 2020-10-05:
ok  in my case, i've deliberately removed pi/Picutres and only want to enumerate the two configured directories.  they could be /home/pi/Lolipop and /home/pi/Pendragon for all I care, my problem is that it's adding back a directory I deleted because I guess it's a default location.  My second problem is just in the formatting of those segments, which I think you are addressing in these code segments, but I am great at ideas and terrible at coding them.

Geektoolkit, 2020-10-05:
Sounds like 2 things:
1) Default directory is getting added back on reboot
2) Need a way to organize the directories so that it's clearer to the user what is a 'top level' and what is a 'subdirectory'

The reason the squares are there is I had this vision of putting thumbnails into each one and trying to make the UI look a lot fancier.  However the directory the images are in is not necessarily one that the user can view from a website, which would require me to create thumbnails for each.   Thinking about Dynaframe long term, I'd like to work to simplify it vs. try to make it too grand, so I'm going to go in favor of something text based.  I'll tinker on this tonight while fixing some bugs and adding some low hanging fruit features.

tmcay's help is great and appreciated, I may move them to a hash table. I've been arguing with myself that the folder structures should likely be objects that store more information and are extensible so that in the future I could track stuff like if a folder has had a file added and such (to let me know when to refresh a folder).  I'll think on that as well.  First I want to crush some low hanging fruit bugs.

qwksilver, 2020-10-05:
1&2 are dealt with (2.06)  adding two new issues created/exposed by the 2.06
3: missing a checkbox for the "root" or first level directory, my testing hasn't yet exposed if photos there are ignored or always used.
4: is is possible to enumerate a few more levels say a max of 5, though personally would use 3

qwksilver, 2020-10-06:
3: thus far not an issue as i don't fill level 1 with photos, just levels 2 & 3.  if I go from pi/home/Pictures i am 4 total levels deep.

Geektoolkit, 2020-10-26:
I'm going to resolve this as not fixable (re: level depth).  I experimented with a few options, in the end they all add complexity.  The initial idea of the folders was to use them as playlists. They were never meant to be used as 'i have a bunch of pictures in these folders'. This provided an easy way to manage playlists...you'd copy images to/from a folder.  If the UI wasn't a webview (and one day it may not be if I write a phone app for this) then this would be trivial and 'safe'. but as it stands currently, there's not a solid way to implement this that won't cause havoc with stability, performance, or both. 

I do have a workaround for some folders...if you say have 2 levels, and want more control over the second, you can add it just like you did the first. This will treat the second level as a seperate playlist, and you can add/remove things as you wish. You can even double the chance of a picture showing up by using this method if you wish. I'm using this for some pictures on my NAS. 

If I find a way to do this in the future I'll revsiit, but I want to close some of these out so I can focus on the most bang for buck stuff, and get to nice to haves later.

closed 2020-10-26 by Geektoolkit