Geektoolkit/Dynaframe3 #18: 2.06 New Items

qwksilver, 2020-10-05:
The changes here are good, but may have generated a few issues:
- [ ] 1: Slow web response, clicking save from the ip interface takes a long time to apply, the page hangs on 
[image]

- [ ] 2: Refresh/Reload page always returns to "controls"  suggest calls line http://ip:8000/settings & /playlist when clicking tabs
- [ ] 3: Suggest Change scaling titles to "Fit Tall", "Fit Wide", "Original", "Fit All(Aspect Ratio Ignored)"
- [ ] 4: Add "Shrink", "Zoom", "Both" option to scalling
- [ ] 5: Advanced idea, can it add blur effect to black space.

to do 5 requires some image processing steps like these:

- 1. Open photo as Fit All and apply a massive blur
- 2. open photo again on top, with scaling parameter,  without first closing blurred layer

Geektoolkit, 2020-10-05:
Yeah, things were bound to break with that many changes, but it gets us to a place where I can actually build something with a bit more of an end goal.  Not sure why it's going slowly, I'm not seeing that. How many folders are you using? I'll point it at more on my end and see if I can reproduce that.

Geektoolkit, 2020-10-05:
I introduced a bad bug where the playlist 'stops'.  I think i have a fix for it, but I'm not clear on how/where it broke.
For scaling, we currently have the following: https://docs.microsoft.com/en-us/dotnet/desktop/wpf/graphics-multimedia/imaging-overview?view=netframeworkdesktop-4.8

The uniform/uniformtofill etc. are what I'm using now.   Also the way WPF scales and I believe the way omxplayer scales may be slightly different. So I'm going to add a video scaling option as well.   But right now it's not doing 'fit tall' or 'fit wide'. It's more 'fit and keep aspect ratio, fit and allow distortion, scale it bigger than the screen but preserve aspect ratio, and 'don't do anything'.

5 is a bit of a larger workitem. I spent 2 days working on that a bit ago to prototype it and wrecked things. I may have a new approach, but for now I'm going to put that as lower priority. 

One thing about this is that I find if you curate the art for it, then it looks MUCH better.  For instance I edit every single photo that goes on mine to match the resolution, aspect ratio, orientation...and then make sure that only stuff that I'd want on a wall as a picture is in the folders.  So anything that scales crappy, or that doesn't fit the screen, I pre-process first.  What this does is makes the project very intentional, and that keeps things looking clean.

I get the desire to have it just work when pointing at an existing art folder, and I think at some point I'll want to handle that.  But I think we should consider if where do we prioritize that...is it over playlist support / better video transitions? Just something to think about as we look at the direction this could go.  I designed it originally to require the curation..and while it's intensive, I feel it rewarding when seeing the results on my walls. 

In the future though, it'd be very powerful to have someone enter a search term and have the frame 'magically' go off and grab artwork using that term and show it, and have it look good, no matter the look. I get the desire of that as well.

Anyway, some thoughts.  I like the tabbed UI, definitely some work to do there. I thought I had it so that when I hit save it would reload to the page that you hit save on. I'll have to try to get a better repro of what you're seeing there.

Geektoolkit, 2020-10-05:
Oh I see the reload bug.  Sorry you explained it clearly, I just didn't think about it that way.  I'll poke around with it...I think if I fix that, any change you make in the UI, and then click a tab...will get saved automatically. Not sure if that is desirable without an explicit 'save' button hit. The other option is we toss your changes when you change tabs unless you save them first.

qwksilver, 2020-10-06:
3: so I tested the scaling with two files, one that was smaller on both edges and one larger.  It really is scaling as described, and in that order.  I've only tested them in 90 orientation.  I think the existing words are ambiguous about what they are doing.  part 4 to add options to only shrink or only enlarge might be nice, but is more bell and whistles type of thinking

as for curation, sure, for the main set sure, but for art collections and other folders, and in my case i could point the frame a a netdrive with 64000+ images.  so having some basic options would be nice.  

eventually I would like to set parameters by aspect ratio of the original, hand have it side or top crop close images, while blur effecting things closer to 1:1 or 4:3.

double eventually it would be great to add a Polaroid transition. where portrait mode could import a series of smaller photos on top of a main photo one at a time at random angles and fixed centers, and scales, to add 4,6,8,9 additional photos on top with white borders, before moving on to the next slide.  and i again i can curate that as an mp4, but that takes away the ability to have that happen randomly and with large chucks of my collections.  

your idea of incorporating and indexing say exif data for tags/keywords takes it up a notch too, so imagine having the functions without needing that advanced curation.  or have it using a netsource like scrollr, tumblr, twitter, insta, and so on.  these things may be a long way off, but laying some good foundation now will save headache later, or heartache over being painted into a proverbial corner.

Geektoolkit, 2020-10-06:
Hmmm. I see what you're saying.  
- I'll get some better wording in there. The current wording definitely is terrible, as I had to look it up myself to see wth it did
- I'm having a lot of problems with the current photo engine and want to rewrite it...making changes is painful because it's all over.  The idea of the poloroid transitions is an entirely different type of 'display' and I could see that being super powerful in the future. I'm going to rewrite the engine and make it extensible. It'll look the same to the end user (well hopefully videos will  fade in but that's tough right now).  But in the end it'll be something that a user can change.

Once I get the tabbed UI working stably with what's there, I think the next thing to tackle is autoupdate.  

So in my head it looks like this:

- Fix what is there...keep the tabbed UI, get it working a bit smoother, redo scaling wording. Try to fix issue #2 above (rewrite the engine)
- Add in autoupdate
- Investigate blurred backgrounds/scaling options
- Other features

For other features I have currently:

- Rclone support - This would be very powerful for screens at realatives houses, or just 'cloud managing' the screens.  I'm even wondering if I could have a generic easily editable config file that all screens would suck down to get the same settings
- Linked Frames - This is a big one for me, as I have 3 now and possibly may go to 4 or 5 soon.  The idea is you'd select a playlist on one, and the others would 'sync' to that playlist. Then you use the webUI for one and it would manage the other 2
- Display Options - Poloroids, other transitions, etc.

qwksilver, 2020-10-06:
Consider creating an engine that can handle all extensions itself.  on PC zoom player can handle a directory full of photos just as well as a directory full of videos, and automatically makes a photo act like a 7 sec video.(or whatever the time out is).  it would be great to pickup webm and gifs in the process.

wouldn't hurt to utilize the pi4's gpu functions.

make the config parse a /home/pi/pictures/settings.ini
and write it all old school dos style, maybe use import/export to put the file where you want it :>
linked frames would be better as separate playlists, but one config. allowing you to log in in one place and update all connected frames.  it would be equally cool to be able to use one pi 4 to run 3 frames via hdmi 1 & 2 and usbc-hdmi 3....using this third screen would require an extra adapter and power through the power pins on the gpio

Geektoolkit, 2020-10-07:
I did some research on webm last night..are you able to play webm files at all on a pi? I keep finding older articles (2016 era) that seem to imply that that hardware decoding doesn't support it, but I also see some mention of it in kodi.  The engine kodi uses is what powers omxplayer, so I'm a bit confused. I don't have any webm files currently, so just curious if you have them and are playing them with performance on a pi, please let me know what you're using. I'll try to get some files when I get back to this (probably a few days away) as well and see what I can figure out. I think I have animated .gif figured out though.

qwksilver, 2020-10-08:
there is a site for plotagraph/cinemagraph that i found.  plotaverse has webm files mostly.
 cvlc --play-and-stop chain.webm :vout-filter=transform --transform-type=90 --video-filter "transform{true}" --crop 16:9
works to play that centered ful lscreen with sides cropped
vlc runs it but still creates and odd error on finish
including one in zipfile for testing

[chain.zip](https://github.com/Geektoolkit/Dynaframe3/files/5345181/chain.zip)

Geektoolkit, 2020-10-08:
Got it, and thankyou....that command line will help me immensely with getting it supported. Also I love the idea of being able to play from plotaverse...that'd give people a ton of content that's animated where this thing really shines!

qwksilver, 2020-10-08:
If you haven't already, look at pi3d for slideshow functions on photos, 
also my VLC media player in PC handles photos in the playlists
also looking into vlc autogeneration of pillar/letter blurs
also will any of this give insight [Discussion on Borders](https://forum.videohelp.com/threads/385341-avisynth-how-to-add-borders)

qwksilver, 2020-10-08:
updated progress on these, also added 6: because
my random is not random across selected folders.  it seems to play a whole folder then move on, also it seems to use the same random order each cycle.

Geektoolkit, 2020-10-26:
I've looked at pi3d, I actually tried to write this in pi3d to start, but it was...messy.  Probably my lack of python experience, but I prefer the land of tight typed class definitions.   That being said, I do love the transitions there and hope to one day bring many of them to this project.

I've been chasing down how to do the blurs, and found out that the framework I'm on may enable it in a version or two.  That feature is moving up on my list of things I want to do.  I also will work on the webm stuff shortly. I did a full rewrite of things and it's MUCH cleaner now so I can actually get bigger items done as long as they don't clutter up the UI too much.  Things like webM, animated gifs should be easier now.  Other transition types that are basic (slides, possibly that poloroid one) may be possible also.  I dont get much time on this, but hoping to get more soon.

qwksilver, 2020-10-28:
everything here addressed or moved to 2.07 open items list.

closed 2020-10-28 by qwksilver