Geektoolkit/Dynaframe3 #19: Clock Display

mulscully, 2020-10-05:
Clock:  Currently my clock displays 19:02 PM not 7:02 PM  allow an option for 12 or 24 hour clock display

Geektoolkit, 2020-10-06:
Will do. There's a way I found to totally allow for customization of the display so that users can choose things like time formats, add in the date, add seconds, etc.   I'll get that added in in the next minor update.  Next major update is going to be to rewrite the slideshow engine so it's a bit more maintainable and sectioned off. A recent change broke it and made me realize I have to do that housekeeping first.

mulscully, 2020-10-06:
BTW,  Thank you for this program.  It is exactly what I was looking for.  I wish I could contribute o help you, but I am not familiar with the framework your using.

Geektoolkit, 2020-10-26:
This is fixed in 2.07. You can how format the clock however you please, including adding in the date, the month, etc.

closed 2020-10-26 by Geektoolkit