Geektoolkit/Dynaframe3 #20: Mixed photo and video playback

keesdekraker, 2020-10-16:
First of all my appreciation for this piece of software. It really works well.
I am running version 2.03 on a RPi3B and can not seems to get a mixed folder to work properly. By mixed I mean both jpg en mp4 files in one folder. It only displays the mp4 file(s) and not the jpgs. When I remove all mp4's from the folder, then all the jpgs are displayed as it should.
Is this the expected behaviour of Dynaframe?

I'm running a clean install of Raspberry Pi OS.

Any help is much appreciated!

Geektoolkit, 2020-10-21:
That's definitely not the expected behavior. 2.03 seems to have introduced a bug. I'm working on a new build which rewrites the playlist engine and adds logging, but it'll be a few weeks before I can get it out.  Please try an earlier release if possible and I'll work to get not only a better release out, but one with logs so it's easier to track down issues like this.

keesdekraker, 2020-10-22:
Thanks for the quick reply. Will try a previous version in the meantime.

Geektoolkit, 2020-10-26:
2.07 should fix this, please try it and leave me feedback. I fixed a number of bugs with video playback in that release

closed 2020-10-26 by Geektoolkit