Geektoolkit/Dynaframe3 #21: 2.07 New items and list of open items

qwksilver, 2020-10-28:
new items
- [ ] a: Bug, visual, minor, on the playlist screen, if there is more than one directory, then the first directory [+] expand marker is offset forward slightly from the rest of the list. 
- [ ] b: Request, setting for the playlist to auto expand, or not when opening playlist screen
- [ ] c: video playback now includes sound, request settings option to mute all video

incorporates list items from other items for links

- [ ] #9 a: **Settings Migration on reinstall**, high priority.
- [ ] #9 b: Black Background slide when video loaded.
- [ ] #9 c: "Rewrite video engine" bugs if found.
- [ ] #18  : Advanced idea, can it add blur effect to black space.
- [ ] #8 a : Misc Comments and issues related to video playback, all low priority at this point
- [ ] #8 b : Misc File Formats to still be addressed (WEBM, GIF)
- [ ] #16 : Multiple Items, reboot, screen on/off, web controls, smarthome calls, gpio calls, etc.

qwksilver, 2020-10-28:
[image]

qwksilver, 2020-10-28:
screen capture for 21e, rotation is my fault, also noted that adding a leading space to the format line fixes this issue and bumps the time over just a bit
[image]

Geektoolkit, 2020-10-28:
Notes:
21a and 21d - good bugs, I see them will work on them.
21b - I'll see what I can do, not sure I can control that.
21c - doh! I'll get an option for that.
213 - Great bug, hadn't tested that extensively, I should be able to fix that though. Also thanks for testing additional fonts, I handn't gotten there yet.  That looks really cool.

9a - Yes Pri 0, the highest. I started work on this today and am having good success. The way it'll work is the first time you install will install 2.07, which will contain an upgrade script which will get copied to /home/pi. From there on you an run that script to reinstall, or pass in a version past 2.07 to install that (so if you want 2.08 you'll run "sudo bash upgrade.sh 2.08" and it'll backup appsettings, wipe out an existing install, download and install 2.08, and restore the settings)
9b - pri 1. Hoping for 2.08.
9c - pri 2, though we may close this. I do now kill omxplayer as you suggested, and I haven't seen the epileptic behavior since I did that. I also do a kill before we ever play a video in case a zombie one goes by.
18 - pri 2.  As a feature I want it pretty badly, but I need to wait for a framework change to implement it.
8a - 2 main issues I see -  Pri 1 (Video copping settings) (Should be easy fix)   Pri3- (Video does not crossfade) 
8b - Pri1 - I want to get that fixed.
16 - Pri2. I have enough for 2.08, lets push these down for now and can revisit.

Upcoming feature that I'm planning:
1) syncing between displays. one would become a primary, and you'd put in the name/ip of others. Then when one changes, it'd  change the others to the same filename. The idea is that you'd point each one to a different folder with a list like 1.jpg, 2.jpg 3.jpg etc. Basically you'd be able to do stuff like this:
https://www.bing.com/images/search?q=5+panel+artwork+star+wars&id=015FFB3FDBCE5BF1E0193BDAAA6AA0B5F78DD158&form=IQFRBA&first=1&scenario=ImageBasicHover
but they'd all change, or animate.

Also: Controls.  I added in arrow key controls in 2.07, not sure if I listed it. The following should work:
Arrow left/right - change forward/back.  It doesn't work 'fast' yet, but it should allow some control
I, H, C, and F - Should info about the file in the info bar if shown (clock must be enabled). Doesn't persist yet.  I is IP address, H is 'hide', C is 'clock' and F is filename (To be expanded up on later with file info)

I'd like to get these into the UI so the web can be used to control the frame remotely.

qksilver you've been amazing and I really appreciate the work and support you're providing!

qwksilver, 2020-10-28:
21b it worked that way when we first switched over to 2.06, then i uninstalled/reinstalled 2.06 and it started this behavior, so that leads me to believe it's possible.

Re listed above list in closest priority order I can. if you give me a list like: <9a,9b,9c,8a,8b,delete 9c, 24, 23, 8, 16,18> then I can set the punchlist to your attack order.

opened #23  #24 to quantify, and discect, items from comments.

qwksilver, 2020-10-29:
9a working with 1 trouble (see end), users need to rerun the 2.07 install process manually.  then later a command
sudo bash UpgradeDynaframe.sh 2.08 will jump to the next version without loss of settings.

It might also be noted that reinstalling 2.07 with the original set of commands detected my existing setup, and saved my settings, reinstalled 2.07, and then created the UpgradeDynaframe.sh for later  at which point the above works for version 2.07

**trouble:**
the "upgrade" script will accept 2.06 as an option. and your settings will stay after roll back.
upgrading back to 2.07 does then cause problems with the web interface, and erase settings. as so.

```
Fatal Error! Excpetion occurred.  Info: System.UnauthorizedAccessException: Access to the path '/home/pi/Dynaframe/appsettings.json' is denied.
```

to fix all this, delete the whole dynaframe, then reinstall with
sudo wget https://github.com/Geektoolkit/Dynaframe3/releases/download/2.07/install.sh && bash install.sh; rm -f install.sh

**recommend saftey to keep from rolling backwards this way.**

qwksilver, 2020-10-30:
Tried update script, had some issues, did clean install, had same issues so here goes, front list includes, but some parts of this are in  #24.

21a, first line fixed, problem now appears as indent on line 3, and double indent on line 4, and 3x-indent on line 5 and so on.  lines 1 & 2 are aligned.

new bug 21f : show time setting no long affects screen instantly, recommend make setting info bar choice: clock, file, version, off, then allow keyboard binds from #24 to change setting, and web interface changes to do the same.

new bug, 21h, a portrait photo, in 90deg, uniform, saved settings.  on reboot those photos have large black bar borders all round, uniform say about 1/2 the screen used by the photo, same space regardless of photo origin.

new bug, 21i, playlist deletions not leaving if checked.  So if you have say pics/landscape with a few items selected, and you "remove" pics/landscape from the list without removing items first, those items are still showing up mixed into the next selections you make, this might have existed in 2.07 or earlier, finding this was an accident

Geektoolkit, 2020-10-30:
re: Update script - I think every time I touch settings, it's going to break now that I think about it. The settings are what gets stored in appsettings, and I suspect if you go backwards you'll always break as there are going to be things it wont understand in settings, and yet it tries to deserialize them.   For going forwards, I think I can fix it. I'll need a few more versions to work through that I think so I can see a pattern of what's happening.  I hit access denied errors despite running as sudo, so possibly something's running still. I'll try to add a pkill in to try to ensure Dynaframe's not running when this is run. Other than that not sure what to do to fix it.

21a..that's frustrating. That took me 2 hours to fix, and I'm still not clear on what I did. I'll try to understand why 3 lines is different than 2.

I'll see what I can do on the next session I can do, likely next week though

qwksilver, 2020-10-31:
21h all of them appear in the same space regardless of sizes, but I'll poll a sample base and maybe make a set to use as testing images. not tonight though.
21a i think can wait for other issues to be looked at, it's basically cosmetic.

re: update script
so i'm thinking back to the days of the .ini file
or a .xml file might be more appropriate these days.

just load a set of variables:
delay=14000
crossfade=2000
rotation=90
showtime=1
dir1="/home/pi/Pictures"

and so on.
then it can load the whole set of variables every reboot,
on initial load it loads the defaults for the version, 
then it loads the ini/xml file iteself, 
then it writes out the current versions ini/xml file whenever settings is saved.

we would get a file to specifically backup and save elsewhere, possibly using a text editor to set the file up in advance of a new deployment.  this would also load the correct set of variables each time after the first run.

qwksilver, 2020-10-31:
21h issues documented in #26 with link to video, tracking relabeled as 26a

qwksilver, 2020-11-08:
added #31, #29, and bolded multi-requested items on master list above.

qwksilver, 2020-11-22:
2.09; 21f, 21g, 21d, all resolved.

qwksilver, 2020-11-22:
closing this because all remain issues are migrated to #36

closed 2020-11-22 by qwksilver