Geektoolkit/Dynaframe3 #23: Future Feature:  Multi Display Synchronization

qwksilver, 2020-10-28:
from comments in #21 
_Upcoming feature that I'm planning:  syncing between displays. one would become a primary, and you'd put in the name/ip of others. Then when one changes, it'd change the others to the same filename. The idea is that you'd point each one to a different folder with a list like 1.jpg, 2.jpg 3.jpg etc. Basically you'd be able to do stuff like this:  [Concept Link](https://www.bing.com/images/search?q=5+panel+artwork+star+wars&id=015FFB3FDBCE5BF1E0193BDAAA6AA0B5F78DD158&form=IQFRBA&first=1&scenario=ImageBasicHover), but they'd all change, or animate._

I would break this down into a few items.
- [ ] a :  Settings to link another frame, ie Main Setting for controller, and linked setting for subordinate units.
- [ ] b :  Timing sending/receiving based on settings above
- [ ] c :  Filename sending/receiving based on setting
- [ ] d :  File selection/display based on matched file, no matching files should mean select random (users that want a piece to be "highlighted" would need to make a blank matched file for these sets)
- [ ] e :  future option: control remote frames playlists via (extra tabs) in the main frame's web ui
- [ ] f :  future option: remote control remote frames  fwd, back, pause, etc as extra lines on the control page

qwksilver, 2020-11-23:
multisync setting tab, by itself for now.
off/on first
is this unit controller yes/no

if no 
ip address of controller
(once controller can be found, disable time/fadetime in settings)
no more options

if yes
number of other screens[}
a list of additional ip boxes for each screen .

for phase 1 send a timing packet, screen number, next slide filename, duration, transition length, so photos all change at same time, and phase in sync.

if working there will be no other functions except all (2,3,4, or 27) screens should phase out and in to new photos at same time.

this much will clear a,b,c above

Spoilers and Complications Below

phase 2 will involve checking that incoming filename in this manner, 
bob.1.jpg  if the .1 is the last thing before extension then next slide will be replaced with bob.<screen number>.jpg 
but only if it exists, if it doesn't then all screens would stay on existing playlists

phase 3 will allow the controller to control the whole web ui remotely so a user only needs 1 controller.
that one controller will have a section for each additional screen on it's web screen allowing

now it's feasible that controlled chaos will be desired, and processing that chaos will be later, but ideas like 
say you have 3 screens controller in the middle then 2 on the left and 3 on the right [2, 1, 3]
bob.t0.2.jpg  ..  bob.t250.1.jpg  ..  bob.t.500.3.jpg
would have the playlist time change extended on screens 1 and 3 by 250 and 500 ms for one slide
phase in left, center, right in sequence but 250ms delay from one screen to next to next
basically including arguments into the filename to allow you to custom curate how things appear and when

another advanced idea to allow screens to always be out of sync by specific amounts. (controller multiscreen settings page) but still coming together in sync and then back to the remaining view time of the interrupted sets before resuming asyncronicity.

qwksilver, 2020-11-25:
[first element of multiscreen started release 2.11 issue comment in release sheet](https://github.com/Geektoolkit/Dynaframe3/issues/36#issuecomment-733987391)

Geektoolkit, 2020-11-29:
A-D are in 2.12. F can be handled using a new browser tab pointing to another frame (and using the control tab inside of Dynaframe)
E I'm going to won't fix...If that advanced of control is needed I'll have APIs that can be used, an automation controller like home assistant can be used, etc.

Please feel free to reopen if you feel strongly about E or F.

I've been wanting to do this for a few months, and so I put a lot of thought into it as I was develping and fixing the last few weeks.  In the end I decided on the following:
1) Multi frame is a bit curated because you have to start deciding what frame is going to control which other frames
2) I wanted almost a 'self healing' situation where if a device dropped off, it wouldnt matter..and if it was restarted or reconnected it'd just be part of the collection.

To do this, I implented it this way:
1) A frame is 'chosen' by the user to be the controller.  You can turn on frame syncing on that one, and then the user should pause any other frame they want controlled (so they don't just go through the playlist on their own).  
2) The chosen frame is provided a list of IP addresses of other frames to sync

If syncing is enabled, as the main frame shows each picture, it uses our HTTP automation channel (same one home automation would use) to 'setfile' to a certain image.  Note for all of below, the image has to already be part of a playlist (Folder should be selected in playlist section).  This prevents anything from coming in and showing an image a frame isn't setup to show (and other badness)
  - If that full path exists...the image is shown on each frame
  - If the folder and image filename exist in the playlist, the image is shown in the frame
  - If there is no folder or path match, but the filename exists 'somewhere' in the playlist, then the file is shown
  - If there is no filename match, but the folder exists...then a random image form the folder is shown
  - Otherwise the frame just shows whatever the 'current image is'.

This allows for three key scenarios:
1) Frame syncing of the same image on multiple frames
2) Frame syncing of parts of an image on multiple frames.  So you could have folders for say 'left, center, right" and then have a file named person1.jpg in each...but the left folder's person1.jpg has the left side, center has center, and so on. This means that one large image can be spread across frames
3) Playlists can be synced.  A 'Scifi' playlist can have images on two frames, and both will show scifi images.  If the sync main frame is set to 'fantasy' and the other folder has a fantasy folder...then the artwork could change to that.  basically playlists would sync up by changing just one frame.
4) If  a person maps NAS shares, uses usb drives, etc. and the paths are different on each frame, but the folder and image names match...everything works as expected.

Closing this as done for now.  There can be some finesses added possibly with things like a pause button next to the frames when they get added. I am torn on adding a fwd/back button next to it...that seems to defeat the concept of 'sync' I worry and could put us in a bad state. I'll think about it.

qwksilver, 2020-12-01:
i'm going to reopen this but reflag it as stretch goals.

I feel pretty strong about E ultimately to be able to command and control the whole "array" from one station is better than multiple logins.  Even if this is a "hack" solution that allows you to jump to other configured frames from the first login.  we'd just need to make sure that there are links on the subframes to return to the main.  

ultimately I would like to see the whole array build new tabs, so the web control screen shows you all frames and gives you buttons, then a playlist tab or column(inside the playlist) for each screen, and finally a setting screen that allows for global and frame specific settings.

for F: half done, we're able to do web controls yes, but more thinking on what they accomplish.
example what if pause is on a monitor and a paired data is received, does the one monitor unpause to show the pair or stay paused. 

pause main if others have matched file shown should pause the set, not just the single frame, or KISS it and just have a pause 1/2/3/all on the main display control screen (more of E)

ultimately a little web styling and a someone good at that stuff could turn the whole thing into a webapp, i don't think a dedicated smart-app is nessessary and have seen plenty that are just containers for an existing webservers content.

qwksilver, 2020-12-18:
new items, i'm not yet fully on 2.14 so some of these may be obsoleted
g:  to change certain functionalities
      -a primary will get to have the list of middles & endpoints
      -a middle is only used to repeat information when on another network, and will also get it's own list of endpoints
      -an endpoint will only receive
h:  the middle and end frames will only activate their own timing and playlist if a new filename is not received in timer+2s
i: this is to help with j, and to establish a name or location name for the frame itself, like Prime(001), Left(002), Right(003)
j: allowing a filename like john_001(prime), john_002, and john_003 and so on allowing one set of shares/syncs to hold all photos
k: i mentioned this elsewhere, but the ability to jump from frame to frame from the main for settings access would be way better than having to manually access and keep track of where you are, to that end having the name and frame designation show in the top bar would be ideal so you always know where you are.

i can see some extrapolations, like each frame having a place to store it's lead frame so you can hop backwards in settings too.

extended example of a middle.  you have a couple locations home and studio  you can use a middle to be the receiver to stay in sync and on theme  at home with the main display at studio, as a kiosk display system you might have building 1, building 2, building 3 all on separate networks, instead of having to  open the router map to all the frames to all the main building you only need one on each network to drive that network.  middle functionality can wait a long time, but should be considered while building the system up.

qwksilver, 2020-12-29:
observations, there really needs to be a subframe functionality, that says while receiving a photo from somewhere else wait x seconds before resume internal ops, and reset that timer each time a new photo arrives.

I'm seeing this timeing, change 1 pic(both screens), 3 seconds later (aproximate boot delay for screen 2 to come online) the subscreen photo changes. and the main doesn't.  it's really obvious here that this is the random/or sequential filename function from screen 2, because it's not nessesarily a second photo from the theme directory.

but if i plug the ip into screen 2 for the main screen, then i get a change at 14sec, and then 3 seconds, and then 11, and then 3, then 14, it's really weird to watch.

right now i'm using the same photo so seeing this change over is obvious.

now it may be a case of having a subscreen come on after a main screen.

qwksilver, 2021-02-06:
closed, moved foward to tweeks in 2.16 #83

closed 2021-02-06 by qwksilver