Geektoolkit/Dynaframe3 #24: Feature Controls, observations and bugs

qwksilver, 2020-10-28:
Migrated from comments in #23 

_Also: Controls. I added in arrow key controls in 2.07, not sure if I listed it. The following should work:
Arrow left/right - change forward/back. It doesn't work 'fast' yet, but it should allow some control
I, H, C, and F - Should info about the file in the info bar if shown (clock must be enabled). Doesn't persist yet. I is IP address, H is 'hide', C is 'clock' and F is filename (To be expanded up on later with file info)

I'd like to get these into the UI so the web can be used to control the frame remotely._
- [ ] a :  hide doesn't stay once selected and revert back to clock/filename on next photo
- [ ] b :  info doesn't stay once selected and revert back to clock/filename on release key
- [ ] c :  bug: larger font size breaks filename display, doesn't fit screen
- [ ] d :  bug: larger font size breaks version & ip address display, doesn't fit screen
- [ ] e :  bugfix: info or filename use fixed default font and size (30), or scaling based on character length
- [ ] f :  arrow keys work but slowly, but will accept multiple presses quickly, need pause/resume with paused indicator.
- [ ] z :  web button bar something simple to access these functions from webui

qwksilver, 2020-10-28:
also just noticed that if clock is not shown in settings, activating it here only shows it for one slide then it also rehides with next photo, all this related to 24a.  24ab might be intended.  i forgot to but it up in the list but filename stays shown in place of clock if you press f.  if clock is turned off in settings, and you hit f to activate filename, then it vanishes with next slide.

i think it might be better to do a few things here
make a setting for filename show noshow
make a setting for info show noshow
and have the buttons c, f, i toggle the onboard setting
and have the H (hide) command work for one slide. to temp hide the other info

have specific locations for each information, like version + ipaddress in small 30% font below clock
have filename show up under those two, in small 30% font below version/ip
and move clock up slightly to compensate for extra lines.
have the clock be persistently line 1 so if it's hidden/off the other two info items still show up in a reduced font size
this would eliminate the "time to show ip address" item in settings since you can turn them on/off at will, since ver/ip/filename are all more or less debugging items having extra font/fontsize boxes are not super needed, and if they are based on a percentage "1/3" of the clock size then a debugger just needs to bump the clock bigger temporarily if they need to read from father away or somethings.

24Z: on the web controls line make a single bar that has icons like: ⏮️ ⏯️ ⏭️ 🕚 ℹ️ 📁 🙈 , that highlight when pause, clock, info, or file are selected..  but hitting pause then hide would let you appreciate that slide without "distractions".  Making it easier to later implement another line for screens 2, 3, and beyond.

Geektoolkit, 2020-10-29:
I like it. I definitely need to rewrite the logic for show/hide to make things more dynamic. This also will enable things like home automation scenarios where we can show alerts (Motion detected on camera) using MQTT, or possibly other notifications in the future. Basically I need a location to put random text fields, and some way to manage it. Ideally if someone marked up image metadata with information such as where a picture was taken, who is in it, etc. I'd like that to be available to show if someone pauses the slideshow.  I believe Mural does something similar with a gesture swipe up today.  I'll think on this and work on it.

qwksilver, 2020-10-30:
2.08, issues 1 & 2 resolved
keys CFIH are now persistent, but not saved across reboots.
clock setting in webui will not bring back the clock if hidden, filename, or ver/ip are selected.
clock setting in webui will not change clock state while live.

Geektoolkit, 2020-10-30:
I'm thinking what this looks like going forward is that the clock on/off setting goes away.  The web UI "control" tab would have buttons to turn things on/off, and then we'd persist that.  The only setting that'd stay is 'show ip for x seconds' so that non techies on first boot would have an easy way to remember / know how to manage the frame.  I'm reading up and thinking on your idea of having the filename/clock/ip as different fields that appear/disappear. 

I'll probably add a pause key, as that should be simple safe, and then try to back off of this a bit and have to prioritize either stabilizing some of the behavior like first run, or adding features and saying 'screw it, after the first image it sorts itself'.

One other idea I have is to throw out all of the 'trying to dynamically change' things on the fly, as that's what is causing much of the odd behavior. In that scene you'd set things in the web UI, nothing would change, but on restart we could cleanly start up because we wouldn't be trying to adjust on the fly. I may take another shot at it, but burned alot of hours and I don't feel like it's much more stable than it was.

qwksilver, 2020-10-31:
this functions are good, especially if later you can start to show specific exif fields like geolocation, author, taken date, etc, but all that is way later, but this is the root of that stretch goal, and will be worth the time in the long run.

say screw it on the image one problem, set the phrase Loading Images to appear exactly center screen, at font size 40 and move on.  the one slide delay isn't the end of the world.

what is bad, I don't think I said in this ticket is that the ip address to not show when set to 0 has stopped working.

Geektoolkit, 2020-11-07:
I'll revisit that logic. After seeing your last video it looks like I hosed that entirely. Something's way off. I agree with writing off the first image and moving on. Thanks for the feedback as always. Excited to get time on this again, hoping this week

qwksilver, 2020-11-22:
24g, 24f resolved, had weirdness in 24f, with first engaging arrow key unresponsive at all then it suddenly kicked in, might be related to vnc connection to send keys to screen, but can't duplicate, so likely not a worry.

suggestions for c,d,e separate font size setting for info, filename, or clock.

Geektoolkit, 2020-11-30:
Closing this one out. Too large of a font size breaking the UI I'm going to 'wont fix' for now. the font size was meant to allow for some readability, but not to get too crazy with it. 

Web bar button is implemented.

closed 2020-11-30 by Geektoolkit