Geektoolkit/Dynaframe3 #28: Wondering about some ideas for remote PI clients...

arretx, 2020-11-05:
I'm blown away by how easy this was to implement.  I had only one hiccup which happened when I fired up the PI with no images in the Pictures folder, but that's it.

So here's the reason I'm hunting for this type of solution.  Christmas is coming.  We just added a little one to the family.  Everyone wants pictures.

I run my own public facing web-server (Ubuntu 18.04) and after seeing Dynaframe, I now how a PI that's delivering images to an old monitor.

I use lightroom to manage my photos, and I publish those photos using the publishing tool in Lightroom to various locations.

If I were to give "photoframes" as gifts, they would need to be configured to auto-update on a regular basis by running rsync over ssh.  Rather than pushing the images to each PI which would require configuring each family's router, I would just bake the configuration into the PI before I gave it as a gift so it would run a cron job and automatically fetch the new images from my server.

That will be the easy part...but it got me thinking about two things.

1.  Would it be possible to provide content based upon conditions, i.e. utilizing Home Assistant at my home location such that it would affect the remote location's image sources?  ...and I think I just answered my own question as I could probably use Home Assistant with NodeRED to manipulate the files in the folders on my server so the sync would update the clients...just an idea.

2.  Layout Templates.  I see single photos rotating on a single screen.  What if we could create a mosaic template, or tiles, or something familiar to users like the Apple TV gallery options so we could display multiple images simultaneously.  This idea stems from the problem of having a vertical photo in the middle of the screen with black sidebars.  If there's a vertical photo, then it could be placed in a multi-photo display template so it can be displayed alongside other photos to fill out the entire screen.

And with that, I'm out...awesome tool!

qwksilver, 2020-11-06:
actually you can do rclone from a dropbox, updating that dropbox updates the frame with a 1min max lag, or whatever your rclone is set to check.

dynaframe knows when the folders change contents and update the playlist accordingly.

arretx, 2020-11-06:
Hmm...trying to avoid using cloud services altogether.  I think I might try running Syncthing on the Pi...or Resilio.  Right now, all of our mobile devices use Resilio to backup our camera roll on a regular basis.  From there I use Lightroom to catalog, tag and publish.

qwksilver, 2020-11-06:
dropbox was my favorite option personally since i already use it, but check here for what all rclone supports.
https://rclone.org/docs/

also this video contains the easiest setup tutorial for rclone, 8min in or so.
https://youtu.be/3GC2EOQbrQo?t=480

Geektoolkit, 2020-11-07:
arretx, thankyou for trying it out, and I'm glad it's working for you. 
Whenever you use the webUI, you'll see that all it does is create a GET request which gets parsed internally to control the frames.  It's something to be aware of..depending on how fancy you want to be.  For instance, you can choose which folder is 'active' using that if you can get an HTTP request to the destination (for families it may be carefully exposing port 8000 via port forwarding).  For your local home, you can easily have node-red send HTTP requests based on conditions (for instance have a holiday folder, and then load that based on the date).  I've used home assistant with V1 to do things like detect my phone and turn on the frame via zwave dongles if I'm home, and show movie posters if my TV is set to netflix.  Using clever folder management, home assistant, and get requests, you can basically have a pretty fancy system (which is what I designed these to be initially...home automation controlled picture frames).  

rclone does indeed support alot, but if you have a syncing mechanism you're probably set.  The latest pre-release has some nice functionality for pointing to differnet folders so you can do things like load from a thumbdrive, a NAS, etc.  I'll hopefully get to work on a release this upcoming week as well. 

I'm glad setup was easy. That was a big goal of mine.  Hoping to make the rest smoother...qwksilver's been great at providing feedback, just need time to work through it :)

qwksilver, 2021-02-06:
1. Rclone, node-red, nas, etc many ways to addreess this.
2. moving to wishlist, avalonia won't support this directly so far, but perhaps an offshoot of pi3d or another software could be called like omxplayer to build it on the fly and then fade away.

closing out, moving to wishlist