Geektoolkit/Dynaframe3 #29: High CPU utilization

racompton, 2020-11-06:
I installed the dynaframe software via the script last night and I'm getting the temperature gage icon in the corner of the screen stating that the CPU is running hot.  I did an htop to show what processes are taking up cpu and I show multiple instances of ./Dynaframe running at 100% cpu utilization.

Geektoolkit, 2020-11-07:
Thanks for this, I'll have to look into why multiple are showing up.  Was this against the 2.08 build by chance?
The framework I'm building on is under development, and has some bad performance bugs currently.  I may have to wait till they release a new build to fix this.   Also can you tell me more about the hardware/cooling you're running? For instance is this on a pi4? is there a heatsink on it, an FLIR case, a fan, or something else helping cool it?

racompton, 2020-11-09:
Hi, no it wasn't on 2.08.  Just the 2.0.3 script you have in the README.md on the main GitHub page.  This is on a Raspberry Pi 3 B+ box.  I reduced the fade time and I modified the RAM assignment to give more RAM to the GPU in the raspi-config script.  This seemed to stop the temp gauge showing up.  
I have no fan or heat sink on this raspberry pi.  Just a case with some slits in it for passive cooling.