Geektoolkit/Dynaframe3 #34: HEIC support?

bobburgers7, 2020-11-16:
Hi,

great tool!  I've been looking for something to use as a family picture frame that combines both video + static images.

Would HEIC image support be something simple to add?

Geektoolkit, 2020-11-16:
I don't believe so at this point. I'm tied to the file formats that Avalonia supports currently, as I'm using their rendering engine (well the entire framework).  I'll give it a shot though and see, and use this issue to track it. 

Other than that, there are bulk tools to convert heic to jpg that may be useful...I know keeping two copies of images isn't ideal, but it'd get them on the frame at least

qwksilver, 2020-11-23:
heic tends to be the new iphone format, and there are mentions of it in avalonia i just can't see where to check for direct support of it, but it looks like it might show up soon.

that said, HEIC/HEIF is also a multi-image/video format, as well as a photo container with crops, color adjusts, transparency, etc. similar to raw, those things will really jack up the performance requirements for the pi running.  slicing out the final image and saving it for screen playback would be ideal

on the flip side a single, finished, photo in HEIC should be 50% smaller at equal or better quality compared to that exported jpeg.  so hopefully avalonia jumps on the train if they haven't already.

As of now this is dependent on an outside library and should be closed here.

closed 2020-11-23 by qwksilver