Geektoolkit/Dynaframe3 #36: 2.09 Open Issue Sheet & Observations

qwksilver, 2020-11-22:
new items
 - [ ] 21i  : playlist deletions not leaving if checked.
- [ ] #8 a : Video Crossfade (kinda awkward since no photo behind it)
- [ ] 21a : Bug, visual, minor, on the playlist screen, if there is more than one directory, then the first directory [+] expand marker is offset forward slightly from the rest of the list.  2.0.8.0 see [comment](https://github.com/Geektoolkit/Dynaframe3/issues/21#issuecomment-719133184) for explanation of new effect.

incorporates list items from other items for links (Ranked by Priority, or close :>)
- [ ] 21e : Bug Font placement not properly accounting for font size, notes below
- [ ] #9 a: **Settings Migration on reinstall**
- [ ] #8 b : Misc File Formats to still be addressed (WEBM, GIF)
- [ ] #24 : Controls, webcontrols, multi items.
- [ ] #23 : multi-display syncing, multi items
- [ ] #16 : Multiple Items, **reboot**, screen on/off, web controls, smarthome calls, gpio calls, etc.
- [ ] #18  : Advanced idea, can it add blurboxing effect to blank space in uniform mode

items in bold have been requested, noticed more than onceshow clock immediately "advances" but resets to beginning of playlist order.

hide info bar immediately "advances" but resets to beginning of playlist order.

note this isn't relevant in "shuffle on", but will start the play cycle over in "shuffle off"

qwksilver, 2020-11-22:
adding to 36a  I've noticed a lot of things, like hitting save on the playlist screen, reverts the playlist to the beginning of the sequence, I've also noticed quite a few times it seems like my "shuffle mode on" just stops being "on", which is what lead to the need for 36b, since we can't tell our current settings from the settings screen unless it's something with a number.

Geektoolkit, 2020-11-23:
Thanks for the quick turnaround, glad I got it back to stable.  2.08 was a huge rewrite to clean things up properly so we could make forward progress.  in 2.08 I narrowed my fixes to just rewriting the infobar code. Thanks also for the suggestion of having it be mode based...that cleaned things up immensely once I let go of the concept of 'clock on/off' and then trying to juggle the rest.

36a - Unless you strongly disagree, going to mark this as won't fix.  To keep the system stable, anytime a change is made on the settings page, I go through and basically 'internally restart' the system.  This means I re-read all of the settings, reset the playlist, etc.  My thinking is that someone should not change settings often, and once they're there, they should stay that way for a bit (outside of the control section to change images on the fly and such).  I'm worried if I get fancy here and try to only refresh the image setup I may jeopardize that.  I'd rather put time into closing down some of the other issues and taking  a look at webm/gif type features.

qwksilver, 2020-11-23:
36a, will become unimportant if we get the web control to select file/clock/info/hide.  the problem in it with reseting playlist is that if you want the filename of the picture you are on, it restarts the sequence before you see the file name.

36c, to clarify this better is not picking up external changes, like adding the new directory on dropbox, have rclone bring it in, but without reboot that new directory doesn't show in the playlist list.

21e, most of this is fixed. if you look at the photo in this [post 21-7](https://github.com/Geektoolkit/Dynaframe3/issues/21#issuecomment-717689444) and see that sliced edge, go on keyboard and go from file to clock to info to wherever and you'll see artifacts that remain on the other size of that sliced edge line.

9, still a fan of settings just being a list of used variables, so that if more is read from the file than the version uses, it's no big deal, and any variable missing is added in with it's default.  ergo load all default variables, load settings variable over those defaults, then resave the settings, then act on it all.  the resave process will ultimately discard anything that wasn't in that version.

8b, will be nice to generate a plug n go environment, but with the tutorials to convert things and make loops, it's not that critical to curate the more important parts of the library at this point.

24, once implemented will automatically eliminate the need to change the 36a behaviour

23, how about starting with a simple multi-screen settings page, with a few things, ideas in a sub-post on #23.

18, fork section and a way to download a running fork for testing, especially if you have most of the code figured out, it may save it in a working form to reference later.

Perhaps go to 2.1 rc-stable, with only the extra web control buttons. (info, file, clock, hide, maybe shutdown now, reboot or restart if feasible at all).  Then begin testing towards 2.2 for another minor rc. and so on.  then 3.0 will be the big reveal point, 

ps if there is a moderator promotion you can give a user, I would be willing to close out duplicates while revising.

Geektoolkit, 2020-11-23:
qwksilver I'm sorry I'm new to github and didn't even think about that! I added you as a collaborator, please let me know if that gives you enough permissions to do what you need. You've been amazing help on this and I'll make sure you're credited appropriately.
I'm reading through the rest of the above.

qwksilver, 2020-11-24:
no worries and yes collab did activate the needed permissions, be warned that it also grants push access, which since i don't code i wouldn't use anyway, but others might and you might want to see what all controls are available there.

qwksilver, 2020-11-24:
2.10 
36a irrelevant now, 
36c resolved with webui update file list button, 
8 delisted may revisit someday
21a delisted may revisit someday
9 i've migrated up two versions from 2.07 to 2.09 then to 2.10, i know issues exist in earlier versions, and in trying to downgrade, 

confirmed to still exist in 2.10
21i unchanged behavior (migrating to seperate issue) now #38 
21e, most of this is fixed. if you look at the photo in this post 21-7 and see that sliced edge, go on keyboard and go from file to clock to info to wherever and you'll see artifacts that remain on the other size of that sliced edge line.

Geektoolkit, 2020-11-24:
I did a quick revisit of webm/gif. Glad to punt those for now.

21i - Tracked this down, will be fixed in next build.
For 21e, I got a repro of the sliced edge...does that stay there for you or is it temporary? I'm seeing it dissapear in about 500ms or so (which would line up with my refresh time).  I'm wondering if it's a rendering artifact, but if it's disappearing for you then I'm glad to wont fix that permanently (mainly because I don't think I can fix it in my code).

Item 9 I'm going to take another stab at before the next release.

qwksilver, 2020-11-25:
2.11 resolves 36b, 
21e deleted as unfixable at this time
fixed 2 unlabeled bugs, in release notes

(36d) setting issues were all somehow around "Should Videos Play Audio (Current value: On )Off" if it's included it doesn't want to save it, I was able to go from off to on, but not back to off.  also settings screen not as fast as before if you don't give it a minute to "save" before clicking on the settings box to open the screen again it tends to scrap quite a few changes.

(36e) new function wouldn't open file then pause, it just always went straight to "paused"  it would do this even if blank after the SETFILE=  NOTE: in my testing I was trying to define a  "/home/pi/Pictures/Friends/Emily/be1.jpg"  and being very specific about the path.  I was never able to advance to an actual picture, i also tried with out the home/pi and with a "../" and a few other varients.

other issues moved to completed side of list to narrow remaining todo list.

Geektoolkit, 2020-11-26:
Ah, ok I had the 'remove settings' bug locked down, then threw in the 'lets show settings for the radio buttons' code. I had everything working well, then hit a point where i couldn't save anything, tore through the code and couldn't figure out what I changed that broke it. I attacked a few areas I thought might've caused it.  Your comment about the audio comment is gold, that will help me figure out what's going on there.

I'm going to add some extra logging around setfile.  I was pausing it on purpose to go into a 'manual' mode but I'm going to scrap that and just have set file set an image, and let the frame do it's normal transitions. I'll have added logging to help track down when it fails.

I actually threw a lot of the sync code in last night.  It doens't take effect unless a synced frame is registered, so it should be mostly harmless. 
2.12 should be good. I'll go track down that setting bug.  I can work on perf after I get the sync settings in.

Geektoolkit, 2020-11-29:
As of 2.12:
#16 - Automation - This is in and done. I need to document it, I'll work on that in the coming week.
#36e - This is working, it's how the backbone of syncing works. 
#23 - Setfile/syncing is in!
#36d - I believe this is fixed, I'm not seeing issues anymore with saving like I was.

I'm looking into blurboxing now.  I have 3 frames synced (I'm using a 65" tv for one of them) and it's been really fun to see. I may have to build wood frames to get them to match up with the TV's depth.

Geektoolkit, 2020-11-30:
#16 - Smart home automation hooks - Now documented in the wiki all commands.

qwksilver, 2020-12-01:
Remaining Issues Migrated to #45

closed 2020-12-01 by qwksilver