Geektoolkit/Dynaframe3 #37: keeps on crashing on RPi4 - logs attached

nash886, 2020-11-23:
11/23/2020 5:25 PM:----------------------Begin Playlist Dump----------------
11/23/2020 5:25 PM:Image:/home/pi/Pictures/a.jpg
11/23/2020 5:25 PM:Image:/home/pi/Pictures/b.jpg
11/23/2020 5:25 PM:Image:/home/pi/Pictures/c.jpg
11/23/2020 5:25 PM:------------------- END PLAYLIST DUMP-------------------
11/23/2020 5:25 PM:Refresh settings was called
11/23/2020 5:25 PM:Current App Settings
11/23/2020 5:25 PM:FadeTransition: 2000
11/23/2020 5:25 PM:SlideshowTransitionTime: 20000
11/23/2020 5:25 PM:FontSize: 50
11/23/2020 5:25 PM:FontFamily: Terminal
11/23/2020 5:25 PM:Clock: False
11/23/2020 5:25 PM:Rotation: 0
11/23/2020 5:25 PM:Shuffle: False
11/23/2020 5:25 PM:VideoStretch: Fill
11/23/2020 5:25 PM:VideoVolume: False
11/23/2020 5:25 PM:Current opacity: 1
11/23/2020 5:25 PM:Beginning to load next file: /home/pi/Pictures/a.jpg
Unhandled exception. System.NullReferenceException: Object reference not set to an instance of an object.
   at Avalonia.Skia.SurfaceRenderTarget..ctor(CreateInfo createInfo)
   at Avalonia.Skia.DrawingContextImpl.CreateRenderTarget(Size size, Nullable`1 format)
   at Avalonia.Skia.DrawingContextImpl.CreateLayer(Size size)
   at Avalonia.Rendering.RenderLayers.Update(Scene scene, IDrawingContextImpl context)
   at Avalonia.Rendering.DeferredRenderer.UpdateRenderLayersAndConsumeSceneIfNeeded(IDrawingContextImpl& context, Boolean recursiveCall)
   at Avalonia.Rendering.DeferredRenderer.Render(Boolean forceComposite)
   at Avalonia.Rendering.DeferredRenderer.Paint(Rect rect)
   at Avalonia.Controls.TopLevel.HandlePaint(Rect rect)
   at Avalonia.X11.X11Window.DoPaint()
   at Avalonia.X11.X11Window.<OnEventSync>b__104_0()
   at Avalonia.Threading.JobRunner.Job.Avalonia.Threading.JobRunner.IJob.Run()
   at Avalonia.Threading.JobRunner.RunJobs(Nullable`1 priority)
   at Avalonia.X11.X11PlatformThreading.HandleX11(CancellationToken cancellationToken)
   at Avalonia.X11.X11PlatformThreading.RunLoop(CancellationToken cancellationToken)
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken)
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args)
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime[T](T builder, String[] args, ShutdownMode shutdownMode)
   at Dynaframe3.Program.Main(String[] args)

nash886, 2020-11-23:
:
sudo wget https://github.com/Geektoolkit/Dynaframe3/releases/download/2.09/install.sh && bash install.sh; rm -f install.sh ; sudo reboot
used this command to install

nash886, 2020-11-23:
Used this command to install: 
sudo wget https://github.com/Geektoolkit/Dynaframe3/releases/download/2.09/install.sh && bash install.sh; rm -f install.sh ; sudo reboot

Geektoolkit, 2020-11-23:
Whoa...can you open a.jpg if you just double click on it? Can you rename it to d.jpg (so that b.jpg loads first) and see if that works. The log indicates it sees the files and is trying to load them, and for some reason crashing on a.jpg when deciding it.

nash886, 2020-11-23:
> Whoa...can you open a.jpg if you just double click on it? Can you rename it to d.jpg (so that b.jpg loads first) and see if that works. The log indicates it sees the files and is trying to load them, and for some reason crashing on a.jpg when deciding it.

a.jpg opens by double clicking. Tried renaming to d.jpg and still no luck..still crashing:
11/23/2020 5:56 PM:----------------------Begin Playlist Dump----------------
11/23/2020 5:56 PM:------------------- END PLAYLIST DUMP-------------------
11/23/2020 5:56 PM:New List generated! Contains: 3 items.
11/23/2020 5:56 PM:----------------------Begin Playlist Dump----------------
11/23/2020 5:56 PM:Image:/home/pi/Pictures/b.jpg
11/23/2020 5:56 PM:Image:/home/pi/Pictures/d.jpg
11/23/2020 5:56 PM:Image:/home/pi/Pictures/c.jpg
11/23/2020 5:56 PM:------------------- END PLAYLIST DUMP-------------------
11/23/2020 5:56 PM:Refresh settings was called
11/23/2020 5:56 PM:Current App Settings
11/23/2020 5:56 PM:FadeTransition: 2000
11/23/2020 5:56 PM:SlideshowTransitionTime: 20000
11/23/2020 5:56 PM:FontSize: 50
11/23/2020 5:56 PM:FontFamily: Terminal
11/23/2020 5:56 PM:Clock: False
11/23/2020 5:56 PM:Rotation: 0
11/23/2020 5:56 PM:Shuffle: False
11/23/2020 5:56 PM:VideoStretch: Fill
11/23/2020 5:56 PM:VideoVolume: False
11/23/2020 5:56 PM:Current opacity: 1
11/23/2020 5:56 PM:Beginning to load next file: /home/pi/Pictures/b.jpg
Unhandled exception. System.NullReferenceException: Object reference not set to an instance of an object.
   at Avalonia.Skia.SurfaceRenderTarget..ctor(CreateInfo createInfo)
   at Avalonia.Skia.DrawingContextImpl.CreateRenderTarget(Size size, Nullable`1 format)
   at Avalonia.Skia.DrawingContextImpl.CreateLayer(Size size)
   at Avalonia.Rendering.RenderLayers.Update(Scene scene, IDrawingContextImpl context)
   at Avalonia.Rendering.DeferredRenderer.UpdateRenderLayersAndConsumeSceneIfNeeded(IDrawingContextImpl& context, Boolean recursiveCall)
   at Avalonia.Rendering.DeferredRenderer.Render(Boolean forceComposite)
   at Avalonia.Rendering.DeferredRenderer.Paint(Rect rect)
   at Avalonia.Controls.TopLevel.HandlePaint(Rect rect)
   at Avalonia.X11.X11Window.DoPaint()
   at Avalonia.X11.X11Window.<OnEventSync>b__104_0()
   at Avalonia.Threading.JobRunner.Job.Avalonia.Threading.JobRunner.IJob.Run()
   at Avalonia.Threading.JobRunner.RunJobs(Nullable`1 priority)
   at Avalonia.X11.X11PlatformThreading.HandleX11(CancellationToken cancellationToken)
   at Avalonia.X11.X11PlatformThreading.RunLoop(CancellationToken cancellationToken)
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken)
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args)
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime[T](T builder, String[] args, ShutdownMode shutdownMode)
   at Dynaframe3.Program.Main(String[] args)

fgf6fqj, 2020-11-23:
I'm on latest OS and firmware:
used this command to get latest updates:
sudo apt-get update && sudo apt-get upgrade && sudo apt-get dist-upgrade && sudo rpi-update

nash886, 2020-11-23:
I'm on latest OS and firmware:
used this command to get latest updates:
sudo apt-get update && sudo apt-get upgrade && sudo apt-get dist-upgrade && sudo rpi-update

Geektoolkit, 2020-11-23:
Ok. I'll work to get a repro. I haven't seen this or had a problem like this. Wondering if I'm missing some recent platform upgrade, it's been a bit since I upgraded. Sorry to hear you're having trouble with it, I appreciate the logs though to help the project out

nash886, 2020-11-23:
Sure no problem.

Geektoolkit, 2020-11-24:
Just an update for my notes: I've done:
sudo apt-get update && sudo apt-get upgrade && sudo apt-get dist-upgrade

with no repro on Raspberry pi OS. I'm not ready to take a firmware update on the one pi4 I have. No repro yet. 

@nash, can you share the resolution of the images, and the source (if known). For instance are they from a SLR camera, downloaded from a website, created in paint, etc.  Looking into the exception it appears that something about the file may be wonky. I'm attatching a sample (png) that is one I know works just to try to see if your setup can render 'anything' or if it's something to do with the files and Skia.  Skia is a rendering engine that is low level compared to where I'm at, so if it's puking on some bits, I don't know if there is much I can do about it sadly unless I can track down what it is about the file that's causing it.
[image]

nash886, 2020-11-24:
Hi @Geektoolkit ,

Tried with your PNG and same error🙁:
11/24/2020 4:18 PM:----------------------Begin Playlist Dump----------------
11/24/2020 4:18 PM:------------------- END PLAYLIST DUMP-------------------
11/24/2020 4:18 PM:New List generated! Contains: 1 items.
11/24/2020 4:18 PM:----------------------Begin Playlist Dump----------------
11/24/2020 4:18 PM:Image:/home/pi/Pictures/100062173-0c794680-2de4-11eb-9cc1-b48f70e93fc2.png
11/24/2020 4:18 PM:------------------- END PLAYLIST DUMP-------------------
11/24/2020 4:18 PM:Refresh settings was called
11/24/2020 4:18 PM:Current App Settings
11/24/2020 4:18 PM:FadeTransition: 2000
11/24/2020 4:18 PM:SlideshowTransitionTime: 20000
11/24/2020 4:18 PM:FontSize: 50
11/24/2020 4:18 PM:FontFamily: Terminal
11/24/2020 4:18 PM:Clock: False
11/24/2020 4:18 PM:Rotation: 0
11/24/2020 4:18 PM:Shuffle: True
11/24/2020 4:18 PM:VideoStretch: Fill
11/24/2020 4:18 PM:VideoVolume: False
11/24/2020 4:18 PM:Current opacity: 1
11/24/2020 4:18 PM:Beginning to load next file: /home/pi/Pictures/100062173-0c794680-2de4-11eb-9cc1-b48f70e93fc2.png
Unhandled exception. System.NullReferenceException: Object reference not set to an instance of an object.
   at Avalonia.Skia.SurfaceRenderTarget..ctor(CreateInfo createInfo)
   at Avalonia.Skia.DrawingContextImpl.CreateRenderTarget(Size size, Nullable`1 format)
   at Avalonia.Skia.DrawingContextImpl.CreateLayer(Size size)
   at Avalonia.Rendering.RenderLayers.Update(Scene scene, IDrawingContextImpl context)
   at Avalonia.Rendering.DeferredRenderer.UpdateRenderLayersAndConsumeSceneIfNeeded(IDrawingContextImpl& context, Boolean recursiveCall)
   at Avalonia.Rendering.DeferredRenderer.Render(Boolean forceComposite)
   at Avalonia.Rendering.DeferredRenderer.Paint(Rect rect)
   at Avalonia.Controls.TopLevel.HandlePaint(Rect rect)
   at Avalonia.X11.X11Window.DoPaint()
   at Avalonia.X11.X11Window.<OnEventSync>b__104_0()
   at Avalonia.Threading.JobRunner.Job.Avalonia.Threading.JobRunner.IJob.Run()
   at Avalonia.Threading.JobRunner.RunJobs(Nullable`1 priority)
   at Avalonia.X11.X11PlatformThreading.HandleX11(CancellationToken cancellationToken)
   at Avalonia.X11.X11PlatformThreading.RunLoop(CancellationToken cancellationToken)
   at Avalonia.Threading.Dispatcher.MainLoop(CancellationToken cancellationToken)
   at Avalonia.Controls.ApplicationLifetimes.ClassicDesktopStyleApplicationLifetime.Start(String[] args)
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime[T](T builder, String[] args, ShutdownMode shutdownMode)
   at Dynaframe3.Program.Main(String[] args)

Geektoolkit, 2020-11-25:
Ok that at least wiped out my concerns about an odd image format. It's tough because there can be so many image types/encodings/etc out there that can throw this for a loop.
I followed what you said and even updated the firmware. After updating the firmware and doing that, OMXPlayer was rendering wonky, Dynaframe would launch to a black screen..the web UI would render, and omxplayer would work, but I couldn't get any images to show up. 

I reimaged back to the latest Raspiban OS, setup wifi, reinstalled Dynaframe and was back up and working. It's a different failure mode than what you're seeing, but I'm wondering if something is getting messed up between Dynaframe, Avalonia, Skia, and Raspberry Pi Os.  My omxplayer playback issues fixed as well after I did that.

I have comments out to the Avalonia group on gitter asking for more guidance. Other than reflashing, I can't do much more at this time with this sadly.

closed 2020-11-25 by Geektoolkit

nash886, 2020-11-25:
@Geektoolkit #

> I reimaged back to the latest Raspiban OS, setup wifi, reinstalled Dynaframe and was back up and working

Did the same and now it just shows blank screen. This is driving me nuts. Tried on RPi3B+ and works flawlessly with no issues whatsoever.

Geektoolkit, 2020-11-25:
What's your open GL settings set to? Can you try turning that off just in case. Also both raspberrypios?

nash886, 2020-11-25:
Both are on Raspberry pi OS: 2020-08-20.

How do I find my open GL settings?

nash886, 2020-11-25:
YAY SUCCESS. Setting to this option in rapi-config advanced settings worked:
[image]

Geektoolkit, 2020-11-25:
Oh man sorry that was such a mess, glad you got it working. I'll get the read me updated immediately