Geektoolkit/Dynaframe3 #38: BUG Playlist "remove" not immediately propagating to playlist

qwksilver, 2020-11-24:
- [ ] was 21i  : playlist deletions not leaving if checked.  part fixed, clicking "remove" doesn't remove included directories from active playlist, however next "save" on the playlist screen does remove those directories.  when you add that directory back, there is no lingering checkbox (unless you didn't ever "save" the new playlist)

Process to duplicate
I have two "roots"
/home/pi/Pictures/Portrait
/home/pi/Pictures/Friends
I check one directory in each, and each has 2 pics, for simplicity slightly different variations of the same person
they are called test and name

now just click remove where the search directory is.
now that whole section is gone from the playlist

BUT it still shows photos from friends even though I removed that whole chain from the playlist.

lets say you don't click save, and go add back/home/pi/Pictures/Friends
and then return to playlist and expand it all and you will still see two "checked" directories.

now clicking save after that delete will actively fix this, but the screen reloaded itself, and when I came back into playlist i didn't know I needed to hit the extra button to kill off directoies no longer available. but maybe the I thought it saved and reloaded so I don't have to

to be fair this senario is me punishing the system a bit, but it might be important to kill off a group and have it gone and not show back up by accident.....

so adding or removing might refresh the file list, but removing doesn't remove from the directories listing, it might be ok for the remove command to automatically do what the save button does too, and this would solve the issue.

Geektoolkit, 2020-11-24:
Perfect, thankyou so much for this. I was able to use this to get the repro, find the issue, and it'll be fixed in the next update.   This also showed me some behavior I needed to document internally better, as the whole playlist code is a bit odd because of adding in the top level files each time...there's two lists I manage..a list of folders under the top level directories and a list of files in the top level directories.  I was only removing the top level directory, and not touching the subfolder list. 
  This was a good cleanup fix, as my remove code wasn't doing anything of use really before. Now it actually goes through and does some proper removal of all subfolders that contained that folder, and does it immediately.  When you remove a folder now it should take effect in 500ms or less (my current 'refresh time'.

qwksilver, 2020-11-25:
2.11 fixes this, and cleans up directory management.

closed 2020-11-25 by qwksilver