Geektoolkit/Dynaframe3 #39: Dynaframe and WittyPi3 conflict

dhariwalm, 2020-11-24:
Hi Joe,
Firstly, thank you for developing such a brilliant program. I am new to all of this and started venturing into the idea for a digital photoframe for the home. Dynaframe has been perfect and works well on the Raspberry Pi3b+'s I have invested in. I wanted to take it a bit further and I came across another RTC power management board called Witty Pi, which connects to the Pi3 and you can manage the start up and shut down of the device. When I have installed this software it conflicts with Dynaframe whereby Dynaframe does not autostart. I have tried re-installing, I have tried the latest versions but all to no avail. I re-image the Pi and Dynaframe on its own is fine, it is soon as I add the software for WittyPi.

Please see the the attached screenshot when I try and run Dynaframe.

I hope this helps, as I would like it to work with the Witty Pi board. If there is something I can amend or if there is any other information you require please do let me know. As mentioned I am new to this whole setup, so my understanding is extremely limited and you would probably have to break it down in dummy terms for myself.

If you can suggest an alternative to the power management board that would also be amazing as I liked the concept of mqtt and I hope someday I understand it well enough to implement at home? 
I did consider changing the cron tab to 'power off' at a particular time but I guess this is where the Wireless plug would come in right?

[image]

Geektoolkit, 2020-11-24:
I think I can fix that. I'll need a few days.

Geektoolkit, 2020-11-25:
2.11 (Being uploaded now) has a potential fix for this, please give it a try and report back. If port 8000 is taken it'll log a comment and switch to 5053.  Appsettings.json allows for further customization of the port if needed.

closed 2020-11-25 by Geektoolkit

dhariwalm, 2020-11-25:
Hi Joe

Thank you for getting back to me so quickly.

Unfortunately the exact same result, I even re-images the Pi for safe measure and Dynaframe works fine as before independently but soon as I add the Witty Pi software, Dynaframe fails to launch.

Even running the script again didn’t do anything.

If there is some information that you need do let me know and I will try and obtain it but it is ok, it was a long shot but I can work just running a cron  job and using the wireless plug to switch off the Pi and monitors

Regards

Mandeep Dhariwal