Geektoolkit/Dynaframe3 #40: 2.11. Some strange looking pictures

Archonw, 2020-11-26:
2.08 scaled all pictures fine (Uniform - Within Screen, preserves aspect ratio)
In 2.11. this happens.
[image]

Geektoolkit, 2020-11-27:
What zoom setting was that on? does changing it affect it at all? In the logs folder should be a Dynaframe.log, if you can find the settings dump in there and post it here that'd help me out.  Also does the second picture also render odd, or just the first (the way the first picture renders is a bit different because it establishes and works from 'nothing' to setup the frame, after that the rest of the pics have at least a reference to go from.

Archonw, 2020-11-27:
> 27.11.2020 10:26:Tried to fix rendering
27.11.2020 10:26:Rotation 90 degrees. calculated  W: 1080 H: 1920
27.11.2020 10:26:Rotation 90 degrees. Panel  Width: 1920 Height: 1080
27.11.2020 10:26:Rotation 90 degrees. Window Width: 1920 Height: 1080
27.11.2020 10:26:Reload settings was true
27.11.2020 10:26:Refresh Directories was true
27.11.2020 10:26:ERROR: GoToNext was unable to find current item in playlist!
27.11.2020 10:27:New List generated! Contains: 6292 items.

Archonw, 2020-11-27:
27.11.2020 10:26:Tried to fix rendering
27.11.2020 10:26:Rotation 90 degrees. calculated  W: 1080 H: 1920
27.11.2020 10:26:Rotation 90 degrees. Panel  Width: 1920 Height: 1080
27.11.2020 10:26:Rotation 90 degrees. Window Width: 1920 Height: 1080
27.11.2020 10:26:Reload settings was true
27.11.2020 10:26:Refresh Directories was true
27.11.2020 10:26:ERROR: GoToNext was unable to find current item in playlist!
27.11.2020 10:27:New List generated! Contains: 6292 items.

27.11.2020 10:33:Reload settings was true
27.11.2020 10:33:Refresh settings was called
27.11.2020 10:33:Current App Settings
27.11.2020 10:33:FadeTransition: 3200
27.11.2020 10:33:SlideshowTransitionTime: 15000
27.11.2020 10:33:FontSize: 50
27.11.2020 10:33:FontFamily: Terminal
27.11.2020 10:33:Clock: True
27.11.2020 10:33:Rotation: 270
27.11.2020 10:33:Shuffle: True
27.11.2020 10:33:VideoStretch: Letterbox
27.11.2020 10:33:VideoVolume: False
27.11.2020 10:33:Current opacity: 1
27.11.2020 10:33:Tried to fix rendering
27.11.2020 10:33:Rotation 90 degrees. calculated  W: 1080 H: 1920
27.11.2020 10:33:Rotation 90 degrees. Panel  Width: 1920 Height: 1080
27.11.2020 10:33:Rotation 90 degrees. Window Width: 1920 Height: 1080

This happens not only to the first picture. I tried different orientations. Only with 270 degrees this happens. All other orientations work fine. 
I think i fixed it by rotating the screen via the /boot/config.txt .

Geektoolkit, 2020-11-27:
ah, rotation 270, this is a dupe of another recent bug found. Ok, good news I have that one tracked down, next release will fix it!

Archonw, 2020-11-27:
Ok, thank you. I love your projekt so much.

closed 2020-11-27 by Archonw