Geektoolkit/Dynaframe3 #42: App for smartphones?

Archonw, 2020-11-27:
Maybe it is possible to create an easy app for Android / IPhone to manage this fantastic projekt.
Browsing via web browser is not really nice. ;-)

Geektoolkit, 2020-11-28:
Tell me more about what you'd want the app to do?

I designed it around the idea that it'd be configured via a webpage, and then it does it's thing. There are http hooks to do most things so that it can be controlled via a home automation system, and there are now keyboard hooks so buttons could be added to control it as well.   But really for my use, I may change the playlist now and then, but for the most part I am not fiddling with it enough to warrant a web app.  I've added the ability to specify paths so the media can point to a NAS or thumbdrive.

Maybe I'm missing something in how it's being used though, please let me know and I'll see what I can do. What I like about the web control is that any development on it benefits ...well everything. PC, MAC, ios, Android. If an iOS or Android developer wants to work on an app I'd be happy to help them. Just not quite sure what it'd do (other than upload photos remotely..that would be a nice feature I'd like to have)

Archonw, 2020-12-02:
Sorry for the delay. 

I just thought it wouldf be nice if i can controll my frame with my smartphone more easily. To open the webpage and try to controll the playlist and the other staff is not very practical on a smartphone.

RichN001, 2020-12-07:
App may not be really necessary, just have to update the webpage to be more responsive. I started playing around with this a lil with bootstrap to make it bit more mobile friendly. 

[image]

RichN001, 2020-12-07:
I imagine we could take care of a upload form too, drop photos in a default Mobile (Playlist) directory or something. Im still trying to figure out where things are and stick with just CSS and html. 

If I wanted to mess with some page settings, I see the stuff in SimpleHTTPServer.cs where does that actually write to on install? I figure I can edit that somewhere for live development.

Geektoolkit, 2020-12-07:
Rich that's cool stuff! I'm way behind on web tech (last I played with it was asp.net MVC days before Razor was a thing). I'll check out bootstrap, that may be the kind of thing that could help me greatly. I've also thought about showing a QR code on the info bar based on a button input so that it's even easier to get to the page.

The reason the web UI is such a pain is that I'm hosting basically a DIY webserver, and I'm swapping out bits on the fly. So SimpleHTTPServer doesn't save what it's doing anywhere...it writes directly out to the client that is making the request.

Client makes http request for default.htm
Simplehttp server sees that request, loads up the default.htm from the web folder, string replaces information, and then serves that resulting string. It's never saved as all of the values are dynamic. 

I thought about moving to a proper webserver, or using flask, or anything else, but any change like that would be massive, and could cause a loss of functionality.  I think the proper thing may be to move to JSON in the app, and then have a seperate webserver that hosts a page that sends JSON to the app to communicate (vs having the app try to do everything on it's own).

I'm honestly not sure what the right approach is here. It's possible that simply making a responsive page using something like bootstrap is the right way to go, as it's possible there won't be a ton of changes needed to the UI at this point. Adding settings is pretty straight forward.   I think it just depends on what does the next 3 - 6 month sprint look like for this, and where the time investment goes. I may need help to really move it forward on that level.

RichN001, 2020-12-08:
I think its pretty clever usage and certainly gets the job done for your needs. QR code is a great Idea. At first I was going to try and make separate pages for each service rather then tabs which is why I was looking in to finding where the simplehttpserver is since that seems where it gets the strings from and I would have to segment that so that it would return to the right page with data. 

Playing with it one page is probably best at this point IMO... going to JSON would greatly improve the future and open up a lot probably but as you said does it need to go to that level, I guess time will tell. 

I have a few things in my head on the UI so Ill play around with it and report my findings.

RichN001, 2020-12-08:
Oh Image upload, if we just have a upload form that allows image or video type supported couldn't we save (http_post) that to /Pictures/mobile dir via your simplehttp service.. basically making a listening class on where to put files. idk, just thinking out loud.. maybe Flask or json haha. although JSON still needs to post to a receiver and in keeping it self installing and just having a web app, maybe a new lightweight web server would be ideal. If we where to develop just a mobile app that pushed json would be different. but I think keeping it self serving would be keeping it to its design. 

Unless you go the route of something like openFrame.

Geektoolkit, 2020-12-09:
I hadn't looked at open frame, that is really cool.  For a second I panicked that I recreated the wheel so to speak, but I think there are things Dynaframe can/will be abele to do long term that will take it in it's own direction.  I do like that openframe can host websites, that is something I want Dynaframe to be able to do eventually, as well as some other ideas I'm working on. 

Yeah, file upload to simplehttp and then saving it to a directory would be ideal.  I just need to understand the API to know what's involved, and if we can do mutli file upload and such.  I am currently investigating some blurring for the backgrounds and webm/gif, but I can add file upload as a backlog item. 

I just verified that 'gestures' work with it by mapping the gesture to a keypress (which DF already supports), and so that feature I'm considering 'done'. I'll document the sensor, but that'll allow buttons and such to control the frame. It's the kind of advanced stuff I've been dying to get to.

SimplehttpServer can be refactored any number of ways. If there is a different way for me to architect that that would help you develop against it please suggest it and I can look into it. One thing to note though...because everything right now is a get request, it makes it SUPER easy to automate the frames using any number of systems. I have home assistant talking to it and syncing it with my zwave lights and it was as simple as creating a getrequest for the frane on/off commands and putting those URLS in.  So there's that benefit of my approach :)

RichN001, 2020-12-09:
Yeah I like openframe but I prefer Dynaframe being self hosted to your frame. Not having to load photos to a web service just to display on my local frames. I could see hosting a server to connect frames to a users frame thou making it a True IOT device, But only making it a option to sign up. Then users could do all the changes via the hosted web app and the server would push to their registered device. I could see that working even with your URL method by adding some auth keys that allow the server to talk to the devices which would be there anyway with registration. You could even push updates... ahhh Pretty endless thinking about it really. 

Ill probably have a new UI working later tonight or tomorrow sometime. I got a few versions I'm working on. So far Im fine with how things are set up.

RichN001, 2020-12-11:
Im going in a few direction trying to figure out the best route. how hard would it be for you to make a single api call and give me all settings in json back. I started working on it but it likes to return to WebTemplate.htm. My theory is if I can make a call and get back ALL current settings. I can then populate check boxes, drop downs etc. Im also thinking about adding multiple frames. So basically you would have a frames section and manually add your frames IP's. 

Yeah Im sorta headed to a actual app as one of my off shoots of doing this haha Im still going to complete my WebTemplate.htm redo but my brain is going crazy on a app..... Do you use IOS or android?

So a request to active_settings.html would give me raw JSON. As I was driving I thought I might be able to do that with some java script at the end of WebTemplate but if its easy for you to just add another page that spits out all settings I thought that may be easier and quicker then me trying to hack a way to get it.

RichN001, 2020-12-11:
A lot of clean up and I would def change design elements on the app, just trying to get some functionality and not spend to much time because its kinda pointless with out getting a dump of settings at this point. You can add multiple frames... then when you select settings, I would pull the settings and populate... then you can adjust like I show in the video and save. Pushing functions I think will be fine the way they are if all I have to do is push a url with them.

The WebApp (WebTemplate.htm)needs some work still including the (hamburger) menu to close when you select on the mobile side but that won't be to hard, then its to the other tabs. I took the current settings that are read in and made them red, I would like to write some java code on that to actually set the fields I think. 

We can work on a logo to or just use text, for looks I added something. 

Multi Frames IOS - http://xgraphics.com/dynaframe/IOSSettingsMultiFrames.mov

WebApp Responsive @ Mobile Device Size (On Desktop the menu runs along the top) - http://xgraphics.com/dynaframe/WebAppSettings.mov

That my update for now.

RichN001, 2020-12-12:
actually if you wanted to make it easier for now probably could just add a field to the current page and hide it. Then in your SimpleHTTPServer.cs you could create a new string replace with json, or even comma separated values for now. 

page = page.Replace("",  
AppSettings.Default.InfoBarFontSize.ToString() + "," +
AppSettings.Default.SlideshowTransitionTime.ToString() + "," +
AppSettings.Default.FadeTransitionTime.ToString() + "," +
AppSettings.Default.NumberOfSecondsToShowIP.ToString() + "," +
AppSettings.Default.DateTimeFormat  + ">"); 
and so on... (I guess values would have to be those that I can pass back with URL, I know you change a few for readability to the user although, I can change them in the app too so raw setting data is fine)

Then in WebTemplate.htm apply to hidden field value.

` <input type="hidden" id="appdata" name="appdata" value="`

With the Web version can we change Replace's for the radios? Then you could set the default values. Rather then text showing the setting. Maybe you tried this and it didn't work Im just talking theory.  Im on a Mac mostly and I don't think I can build this or I would try. I guess I could build at work thou so if you don't have time to mess with this now I totally understand. Ill try to set it up on Monday and make some changes. The above is a quick way that I could possibly get data to use on a app and the below changes hopefully would work on the current version. 

example if we setup a If statement on these we could use the checked variable on html and just replace the actual value. I guess this could get involved with replaces as there would have to be 2 options per radio item and we would have to replace all values in html. 

for rotate in SimpleHTTPServer.cs.. (just in text terminology..lol)

```
0True  = value = "0" checked> 
0False = value= "0">
= <!--ROTATIONCURRENTSETTING0-->>

90True  =  value = "90" checked> 
90False = value= "90">)
= <!--ROTATIONCURRENTSETTING90-->>

```
etc....

In HTML 

```
<!--Rotate-->

	<div class="form-label"><span><strong>Rotate </strong></div>
	<div class="form-check form-check-inline">
  <input class="form-check-input" type="radio" name="rotation" id="radio-0" <!--ROTATIONCURRENTSETTING0-->>
  <label class="form-check-label" for="radio-0">0
</div>
<div class="form-check form-check-inline">
  <input class="form-check-input" type="radio" name="rotation" id="radio-90" <!--ROTATIONCURRENTSETTING90-->>
  <label class="form-check-label" for="radio-90">90
</div>			
<div class="form-check form-check-inline">
  <input class="form-check-input" type="radio" name="rotation" id="radio-180" <!--ROTATIONCURRENTSETTING180-->>
  <label class="form-check-label" for="radio-180">180
</div>	
<div class="form-check form-check-inline">
  <input class="form-check-input" type="radio" name="rotation" id="radio-270" <!--ROTATIONCURRENTSETTING270-->>  <label class="form-check-label" for="radio-270">270
</div>	

```

RichN001, 2020-12-12:
Actually it sorta builds... ill work on it later.. Kids are irritated at me haha.

Geektoolkit, 2020-12-13:
Sorry Rich was taking care of some life stuff. I'm catching up on this now. I think sending the settings down on JSON is doable, I'll have to see what I can do.  It may build on a mac actually, it's written in .netcore with avalonia which is meant to be cross platform.I just dont have a mac to test it on

RichN001, 2020-12-13:
all good.. yeah i got it to run on Mac, there where a few issues but it runs. certainly enough for testing.

RichN001, 2020-12-13:
ok fully running no issues at all. Just had to update some things runs perfectly on Mac....  :)

Geektoolkit, 2020-12-13:
What's the command line to play a video full screen on a mac? I can get that added in. If there's a rotaiton option I can add that as well. I thought about adding a video player .exe separately for windows, but it may be easier for windows users to just rotate the screen.

RichN001, 2020-12-14:
i'll have to go experiment. Honestly i've only ever used VLC Player which you can do mostly anything from command.

RichN001, 2020-12-14:
Not sure what I did at home compared to the office but not having as much luck here building on Mac at work. I compared Dependencies everything seems ok.... Then I tried on Windows box and im getting System.InvalidOperationException: "Sequence contains no elements" and stopping at CurrentPlayListItem = CurrentPlayListItems.First(); on both so must be something with my setup on Mac and pc. Ill look back in to in a bit.

Geektoolkit, 2020-12-14:
make sure there is a picture in the picture folder! I need to fix that

RichN001, 2020-12-14:
lol Ive been going crazy here haha.. thanks.

RichN001, 2020-12-14:
Mac Built ok, Windows stops on videoProcess.Close.  System.NullRefrenceException 'Object reference not set to an instance of an object'... Just trying to get them both going so I can compare the runs. any ideas?

RichN001, 2020-12-14:
For your Mac video question, Quicktime is the default player. You can start a mov from command but there is no switch for full screen or rotate. However I could write a apple script that you could run that would do it but may be better off using VLC and making a prerequisite I guess, but all the options are available for command line.

RichN001, 2020-12-16:
I have got json with current settings going to a hidden field for now from the default page. I also redid somethings to make the radio buttons set themselves to the current settings. This is for the current version page my redo of that later... Ill commit on my fork and push for you to check out later. Accept it if your cool with it.

Geektoolkit, 2020-12-16:
I'm going to look at VLC performance on the pi.  It may be really slick to move to VLC and then have a consistent video playback experience across mac/windows/linux which would be nice.  The last I checked VLC wasn't as performant as omxplayer on the pi3B, but it's been a bit.  If nothing else I can add a hack like I did for windows which works in landscape only, but  can point to VLC to at least provide a solution.  in windows i hardcode it to a default windows media player path and that actually works ok as long as I am not trying to fade/rotate it.  

The PR looks good, i want to go through and test it properly so it may be a day or so but I'll get it in. Thanks for fixing the radio buttons, that makes it alot nicer. I'm going to have to see what this JSON string does so I understand it better. 

For Windows, make sure you run Visual studio Elevated. I think the httpserver may cause issues otherwise. I'll get some notes into the wiki for development.  

I also need to add a check for 'no pictures in picture folder' and set a custom infobar message. :)

RichN001, 2020-12-16:
Sounds good... I have some other questions about natively running. that's why I was trying to get windows going to.. Ill mess with that. This way I can compare for you and tell you what's different.. etc. Should we start a new discussion for compatibly, email you.. or just leave it here?

when you get to the review... page source on the html will show the below. I was ok with the raw info and did not need fancy names as I can do that on parse. any way.....  back to working on a more responsive html page. I wanted to get those radios working it was driving me crazy..lol... 

					
					<input type="hidden" id="appdata" name="appdata" value="{
  "Shuffle": false,
  "VideoVolume": false,
  "ExpandDirectoriesByDefault": true,
  "Rotation": 0,
  "OXMOrientnation": "--orientation 0",
  "SearchDirectories": [
    "/Users/rnewberger/Pictures"
  ],
  "CurrentPlayList": [
    "/Users/rnewberger/Pictures/Photos Library.photoslibrary",
    "/Users/rnewberger/Pictures/Lightroom Library.lrlibrary"
  ],
  "InfoBarFontSize": 30,
  "InfoBarState": 0,
  "FadeTransitionTime": 1600,
  "SlideshowTransitionTime": 15000,
  "DateTimeFormat": "H:mm tt",
  "DateTimeFontFamily": "Terminal",
  "CurrentDirectory": "/Users/rnewberger/Pictures",
  "NumberOfSecondsToShowIP": 10,
  "ImageStretch": 3,
  "VideoStretch": "Fill",
  "ReloadSettings": false,
  "RefreshDirctories": false,
  "ListenerPort": 8000,
  "IsSyncEnabled": false,
  "RemoteClients": []
}"

Geektoolkit, 2020-12-17:
PR Approved, it'll be in 2.14 (hopefully coming out tonight). The radio buttons are beutiful. Sometimes I get code paralyzed thinking 'there's gotta be a better way' but the way you did it was correct and elegant. I think whenever writing out a page like I am things are going to be a bit messy/verbose. I like that you pulled the JSON directly from settings, that means I don't have to worry when updating the appsettings file, which gets a ton of churn as features come in. 

My email is joe farro at hotmail dot com without spaces.  Feel free to open a new discussion or email me there, either works. 

2.14 is uploading now, but very little testing at this point.

RichN001, 2020-12-17:
Thanks glad I could contribute... haha I know what you mean, I did it 2 times then was like "O" I should just do this and re-did it all. Yeah seems like the best way to get the Settings, It works for now and its not a big load of data so it doesn't hurt anything. Id like to get that on another page (appsettings.html) at some point and possibly look at putting a key on the request. That would limit the call to only when needed rather than every time the settings page is used. Like I said it's not much data now but who knows what the future holds. 

Also leaves the fact that having a frame on public net would help secure the settings, but that also means we would need to put some password management on the default page and include the 'key' in the update of settings with your current method as well to help lock that up. Down the road haha... 

At least now when I get more in to app dev I can pull that json. ;) 

Ill email you a few things with some preliminary questions then Ill probably start a Mac thread.

qwksilver, 2021-02-06:
i'm closing this since most of this is implemented, and issues will be separate, but i will leave this conversation listed for reference in the 2.16 compilation post.

closed 2021-02-06 by qwksilver