Geektoolkit/Dynaframe3 #43: WebUI / Image - Questions and Issues

zackpelka, 2020-11-28:
I finally got the image to work on my Pi 4, I think I had to initially plug drive into a Pi 3+, kill the install (turn it off) and then threw it into the 4. Blamo, now it works. 

But...
When I access the WebUI manually by clicking on the Webtemple.html file, it loa/ ds the UI, but throws a 404 error when I try and hit save. The HTML submit I guess gets passed to the js files or something right? Is the file missing and needs to be created with +w perms, and if so, what is the file name?

Also, from an iPad, how can I access the Webui without going in manually to the console for Pi?

zackpelka, 2020-11-28:
ok, by copying the Webtemplate.htm to default.htm, it's now posting it to itself. 1 down... 

But I still need to know if there are keyboard shortcuts to get to the WebUI?

Geektoolkit, 2020-11-28:
The app hosts a webserver, so the web UI can be launched from anything in the same network. It's not meant to be launched manually. Http://yourip:8000 and you should be up and running! I use my desktop to configure my frames

Geektoolkit, 2020-11-30:
this has been added to the wiki. Closing..but if you have any issues I"m monitoring and can help.

closed 2020-11-30 by Geektoolkit