Geektoolkit/Dynaframe3 #44: video playing not working

Techserv-krY, 2020-11-30:
Hello 
Nice Project!

The videos are not played.
Tested with 2.09&2.10.
Extract from the log file
OMXPLayer args: --orientation 0 --aspect-mode Fill"/home/pi......
There is no space between Fill and "/folder, that could be the error.
System: Raspberry Pi 4
File format: .mp4

Geektoolkit, 2020-11-30:
Ah, good bug, I see it. I'm guessing you have the video volume set to 'true'?  Can you try setting that to false please and see if you repro it (or opposite of where it's at.

Next build will have a fix for this. Thanks for reporting it!

Techserv-krY, 2020-11-30:
video volume is true for this error.
changing to false it does not overwrite appsettings.json
even if I change the file manually it will be overwritten to true

Techserv-krY, 2020-11-30:
when I click on video should play audio "off", it will be checked the "off" for shuffle off and not for audio

Techserv-krY, 2020-11-30:
WebTemplate.htm ther is the issue 
`<div class="form-label"><span>Should Videos Play Audio </span></div>
						<div class="form-input"><input type="radio" name="VideoVolume" id="VideoVolume" value="on" /><label for="shuffle">On</label><input id="shuffle-off" type="radio" name="shuffle" value="off" /><label for="shuffle-off">Off</label></div>`

`<div class="form-label"><span>Shuffle </span></div>
						<div class="form-input"><input type="radio" name="shuffle" id="shuffle" value="on" /><label for="shuffle">On</label><input id="shuffle-off" type="radio" name="shuffle" value="off" /><label for="shuffle-off">Off</label></div>`

Geektoolkit, 2020-11-30:
ah right, I fixed that in 2.12.  Stupid copy/paste issue on my part.  A workaround for your build would be to append the following to the default.htm:
?VideoVolume=off

That should get it to go to off. 2.13 will have the true fix for this (proper spacing of the command line).  Hoping to get that out about Wednesday, working on some cleanup and the wiki/documentation in the meantime. If you need it sooner let me know, I appreciate the feedback!

Geektoolkit, 2020-12-06:
Thanks so much for opening this issue and helping me pinpoint it. This should be fixed in the 2.13 prerelease.

closed 2020-12-06 by Geektoolkit