Geektoolkit/Dynaframe3 #45: 2.12 Open Issues Sheet

qwksilver, 2020-12-01:
- [ ] #8 b : Misc File Formats to still be addressed (WEBM, GIF)
- [ ] #18  : Advanced idea, can it add blurboxing effect to blank space in uniform mode

Geektoolkit, 2020-12-06:
#44 resolved in recent update (2.13)

qwksilver, 2020-12-08:
is the file sort currently full filename, like path and all, or is it just the filename part?

Geektoolkit, 2020-12-08:
I believe it's just filename currently. Code looks like this:
 CurrentPlayListItems.Sort((y, z) => new FileInfo(y.Path).Name.CompareTo(new FileInfo(z.Path).Name));

The good news is, now that I have that, I should be able to sort the entire list by date, path, filename, etc. as long as it's the entire playlist.

I could see the following options being provided/useful.  "Overall" would be across all playlists/folders.
Alphabetical per folder
Alphabetical overall
ByDatePerFolder
ByDateOverall

And then maybe a second option for ascending/descending.  

Let me know if that seems right, or if I'm missing scenarios there. Now that I have an intitial sort algorithm, I think it should be fairly straightforward to add.

qwksilver, 2020-12-09:
I'm onboard with those making sense
there are often multiple date attributes, so it might be date taken, date modified, date created, or something like that to work with as well.  I can certainly see scenarios where you might want it by dake taken/created or by date added (modified).

and definately on ascending descending.

qwksilver, 2020-12-11:
2.13.0.0 I'm getting weird effect on the settings screen related to #44.  Clicking off on video is turning off shuffle after save.  You can kinda see it on screen with the selector boxes too, clicking on shuffle, followed by off video causes the (on-suffle-bubble) to vanish, and likewise reclicking the shuffle on box causes the (video-sound-off-bubble) to vanish.  I've also only been able to disable video using the direct command line command.  

on an unrelated note, I saw the update video it was great!  and you really need to [hide that wiring](https://www.lowes.com/pd/MonoSystems-4-Piece-4-in-White-Flat-Screen-TV-Kit/50053817) in the wall!

Geektoolkit, 2020-12-11:
Damn. Yeah it's still regressed in the webtemplate.htm...the audio is affecting shufffle.  I see the bug, I can get it fixed next build.

Thanks about the video, spent alot of time getting that going.  The wiring ...omg yes...it's a project though. I want to swap the 22" frames out for 32" ones on either side...then I am going to run the wires from the side frames across the back of the TV (which will get rid of the wires going downward).  Then I'll have a single bundle to run down..and I need to decide if I bring power up to a recessed box, or do an in wall conduit style and run it down to a single 'out' section at the bottom. So it's all in the plan, but taking my time because it's a new house and if I cut into the wall, I want to be super thoughtful and make sure I don't change my mind in a year.

qwksilver, 2020-12-12:
I would use in wall rated extension cords, if you have 3d printing, then print an appropriate case, i've seen a pi-hat for PoE, then run 10 ft pi hdmi cables and put all the pi's and cooling and such at the base of the wall, 2 10', 1 6', and 3 powers in a 1g cut in box.  If we are talking an inside wall, you can do it all with this list.

[article 1:  ](https://hackaday.com/2019/09/30/using-poe-with-a-raspberry-pi-3-for-about-two-bucks/)  If you want to try for a hacked power over ethernet system
or[ 3 cables from this amazon(2 10', 16') ](https://www.amazon.com/BlueRigger-Speed-Micro-cable-Ethernet/dp/B0060AU5MW/ref=sr_1_8?crid=15PNYWANGKUBC&dchild=1&keywords=microhdmi-hdmi+cable&qid=1607734172&sprefix=microhdmi%2Caps%2C661&sr=8-8)
[a couple of these, one goes level with other electrical outlets](https://www.lowes.com/pd/CARLON-1-Gang-Orange-Plastic-Old-Work-Standard-Rectangular-Wall-Electrical-Box/50385688)
[a good in wall extension to power tvs](https://www.amazon.com/SIMBR-Extension-Angled-Triple-Outlets/dp/B071HFSSXR/ref=sr_1_7?dchild=1&keywords=15+amp+extension+cable+in+wall+triple&qid=1607734454&sr=8-7)
[and these to pretty things up, especially when reselling house later](https://www.amazon.com/Fosmon-Opening-Passthrough-Voltage-Installation/dp/B072L2J5DF/ref=sr_1_4?dchild=1&keywords=cable+access+plate&qid=1607734809&sr=8-4)

Inside wall with no pipes, insulation, bulkhead, or blocking and its a really easy setup and the wall mods themselves take about 20 min.  if you have an outside wall, or otherwise a wall that is insulated, then you need a pullrod to help, but otherwise no big change.  I always do the bottom one first and then look up the wall space to make sure nothing is in the way.  

I just sent a fr if you want to dm me for any more info on it, I spent just over 8 years working as a professional electrician, and while the tech is newer now, it's just fantastic what can be done compared to then.

qwksilver, 2020-12-28:
44, 49, 51 should be resolved in 2.14 or 2.15
moving the rest to a new issue

closed 2020-12-28 by qwksilver