Geektoolkit/Dynaframe3 #46: Not All Pictures Displaying + Sequence is Pseudo-Random...

cotedelechet, 2020-12-03:
Joe, Great job (& video) with this software.
I'm running Dynaframe on Pi 3B (1GB).  I have ~50pix + 5 videos (.mp4) in 1 folder only.
The pix are between 4-8MB in size.  I've turned Shuffle=Off in the Settings.

The pictures & vids are displayed in a pseudo-random sequence, not by filename or (modified) date.
Additionally, not all the pictures get shown, only about 12 of the 50 pix in the folder get displayed over & over again, in a not so random sequence.

Is there any limit to the picture/video filesize ?? or does the Pi 3B not have enough memory ?
If Shuffle=Off, what is the default sequence for displaying the files ?? Can I sequence by filename or mod date ?
Thanks, DL

Geektoolkit, 2020-12-04:
Thanks for the feedback and the issue. This could be quite a few things.  I had some bugs in earlier versions around the playlist playback.  Please try 2.12 if you're not already and let's see if you aren't seeing all of the images still. 

The other thing is there is a log file that gets spit out (or if you run it manually it goes to the console).  If you can share that I can make sure that settings are getting applied correctly and such. 

There is a resolution limit as far as a pi3 and memory are concerned. I've done some images that were 4k resolutions or higher and it's caused issues.  The 2.12 version is much more efficient with memory and CPU so it may help there. 

The sequence thing isn't one I've thought about deeply.  I'll think on it and see how tough it'd be to do a sorting based on filename or date.  It can be tricky when there are many folders because then I have to know if the intention is filename by foldername, or filename despite foldernames, or provide options for both. The matrix gets big quicklly, so I have to be thoughtful of how to add it.

cotedelechet, 2020-12-04:
Joe, Thanks for the quick reply.
I've installed v2.12.

Dynaframe still does not appear to display all the pictures in the folder
-- it seems to display a subset of the pictures over & over.
Attached is the log file.

Also, several new issues with v2.12:
1.  As soon as I move my mouse, the screen goes back to the Pi Desktop.
Dynaframe still appears to be running (ps -A).
2. v2.12 does not play any movies (worked in v2.09).

Suggestion: on the web interface (Welcome to Dynaframe!), I would suggest
showing the (current) software version somewhere.

Regards,
D Lee
ps.....I need to get the pi 4B

qwksilver, 2020-12-04:
I've noticed this more while playing with the files, I'm fairly sure the random sequence rebuilds after there is a change to either the web interface settings, or to the contents of the files directory.  Depending on how the rng is seeded you would likely see a lot of the same files early on.  I you haven't already see if the system is including different content after the first hour or so.

many programs driven by a randomizer seem to have the issue of large chunks of "preferred" files.

as for sort by filename, sort by date(directory), sort date directory(random), sort by date full library, and so on.  all the options but more on that later, I'm about out of time for today.

I think early on there was also a suggestion that new photos added have some sort of priority in the playlist for a bit, if that happened, and there is code for that, it might be affecting the playlist generator.

Geektoolkit, 2020-12-04:
The big where you move the mouse and the desktop appears...is that when you're using VNC? I believe that happens with VNC only. I hope.

cotedelechet, 2020-12-04:
No, I'm not using VNC (I'm booting straight into the Pi).

DL

Geektoolkit, 2020-12-04:
Ok. I can look at the mouse movingment as a separate, lower pri issue. maybe has to do with unclutter.

The attatchment didn't come through, can you please try on github (it looks like it tried via email). Or if it did I'm not seeing it. The main thing though is to see if the playlist in logs matches the files...if so, then DF is picking up all of the files and not going through them properly.  If the files aren't showing up i the logs, then that makes me think something else is up.

I didn't do a new photo priority algo.  The way it's coded it looks at the top level folders, then scans the directories..and for any selected in the UI it gets the files. It adds them t a 'playlist' and then starts at the top and works down.  If it gets to the end, it goes back to the top. If the webUI directories are changed, it refreshes the playlist.  

If 'shuffle' is enabled and the save button on the UI happens, it shuffles the playlist and goes through it again.  When its done, it reshuffles it and goes through again.

It's possible we're not picking up top level folders.  If we can figure out what in particular about those files (extension, resolution, folder, etc) it'll help me out.  I'll not get time to look at this in code till next week.

cotedelechet, 2020-12-05:
Joe, here's the DF log.
[dynaframe.log](https://github.com/Geektoolkit/Dynaframe3/files/5646913/dynaframe.log)

Geektoolkit, 2020-12-05:
Ah! Ok thank you, please set the "play audio" setting for video files to "off". That should fix videos (known bug, fix is coming in the next release).

I'll review it more later to understand the ordering and try to figure out why the file list is getting cut off

Geektoolkit, 2020-12-06:
Ok I had a chance to look at the log. I see it saying it sees 58 items. 2 of those are videos which aren't being played because of a known bug when audio 'on' is set.  The other 56 it seems to be iterating through correctly.  Do you have more than 56 image files (and 58 total files) in Image:/home/pi/Pictures/Norfolk-20160813 ?   If so please let me know, and if they have any extenions outisde of jpg.

The good news is I found out how to sort by filename pretty easily, as well as sort by date (and reverse sort).  I should be able to add that in as an option in an upcoming build.  For now I'm going to target just sort by file name across all folders for 2.13.

Geektoolkit, 2020-12-06:
Please try the recently 2.13, I believe this is fixed in there. The playback issue with omxplayer is fixed, and I have a setting to make all files play in an 'alphabetical' order. If that doesn't resolve it please reopen.

closed 2020-12-06 by Geektoolkit

cotedelechet, 2020-12-07:
Thanks again for the quick response & software updates.
I'll install v2.13.

*Do you have more than 56 image files (and 58 total files) in
Image:/home/pi/Pictures/**Norfolk-20160813 ? If so please let me know, and
if they have any extenions outisde of jpg.*

There are just those 56 .jpg + 2 videos (.mp4) in that 1 folder.  There are
no other folders, files/file extensions.

Thanks, DL