Geektoolkit/Dynaframe3 #47: Feature Request: Video duration playback

nesstea, 2020-12-03:
Hi. I was wondering if you could allow a setting to cycle through the media files as normal, but enable the duration of the video to be played, and then cycled to the next media file.

Example:

show picture every 10 seconds. once a video is being played, if option is set, get video length and play that video for that duration, then move on after a second of transition time.

This would greatly increase the value of your software.
Thank you.

Geektoolkit, 2020-12-04:
It's a good request, and one I can work on when I get back into this. Likely not until the new year. The engine is capable of this though. Right now I have a few other requests like blurboxing that I'm looking into, but need to take a break for the holidays.

rpluijmen, 2021-04-29:
I created a local branch for this feature. However I would like to push it to a branch to create a PR. I don't have access (403). Is that something you can provide? 
[image]

Geektoolkit, 2021-04-29:
Awesome! I just invited you in as a collaborator, please accept that and you should have permissions to get that in

rpluijmen, 2021-04-29:
Thanks. I created a PR. 
I tested this on my local Windows machine and a raspberry PI 4 and it both worked. Let me know what you think.
If you have any questions just let me know

Geektoolkit, 2021-06-27:
Closing, this is in recent builds, and when stabilized this will be available to all in the next release (post 2.23)

closed 2021-06-27 by Geektoolkit