Geektoolkit/Dynaframe3 #48: Feature suggestion: COMITUP

RichN001, 2020-12-07:
Hey, thanks for spending the time on this its come a long way since I last looked at. Just a idea to maybe include Comitup in install... this way if WIFI is down or changes, the frame will start a Access Point on boot so that you can configure WIFI from a mobile or other device rather then have to get a keyboard and mouse hooked up. 

Easy enough to install just thought it would be nice for a Zero config frame or a work in that direction. 

https://github.com/davesteele/comitup/wiki/Installing-Comitup

Geektoolkit, 2020-12-07:
Yes!  Thankyou so much for this, it's something I've wanted to solve and didn't know how to setup a captive portal myself.  That looks like exactly what I'd want to do. I'll look into implementing this, as this would help usability greatly, and possibly enable a mode where it doesn't need internet connectivity at all! (which I'd prefer as an option for secure installs).  Thankyou!

RichN001, 2020-12-07:
sure.. it works flawlessly. I like the idea of not having to connect the frame to the network too. You can change the AP SSID, but might have to get a update in there on boot to set a variable like a UUID. Ex... Dynaframe_AP_UUID that way each frame would have a unique SSID by default. Then maybe Add something in settings for custom SSID name.

Geektoolkit, 2021-06-27:
quick note I spent some time on this. Part of installing it it actually takes the pi off of the network and you have to delete some of the wifi files to let it control the connection. I need to think on it more of how I'd install this and not knock someone's internet connected pi off the network  in such a way that would be disruptive.

Geektoolkit, 2021-09-01:
I've investigated this, and for COMITUP to be released, it'd be a much better image based solution than having dynaframe try to do it as is. I want to keep this build as something someone can add to a machine without it taking it off the network. I have plans of a dynaframe pro that will use comitup, and i'll move this to that project.

closed 2021-09-01 by Geektoolkit