Geektoolkit/Dynaframe3 #49: Gets stuck cycling 3 images after a few hours

driftinaway, 2020-12-09:
Hello. First time posting here so apologies if I have this in the wrong spot.  I recently loaded version 2.12 onto a new Pi 4.  I purchased it just for this project, loaded raspbian OS - updated everything then loaded Dynaframe.  No real issues loading or getting it up and running - appreciate the great info and video. Overall exactly what I am looking for but after a few hours it seems to get stuck repeating only a few images out of a larger list and basically stutters. I adjusted some folder names and file names(getting rid of spaces etc) as a first step but no change.  I included a section of the log file below. I kicked it off and watched it cycle through all pictures in the playlist a few times no issues. 3.5 hours later at 6:40 it was stuck repeating the same 3 images

[dynaframe_log_Section.txt](https://github.com/Geektoolkit/Dynaframe3/files/5662714/dynaframe_log_Section.txt)

qwksilver, 2020-12-09:
Please install 2.13 and let us know if this issue still exists, a bunch of work on file handling was part of 2.13, and may have fixed this issue.

see here for latest versions:  https://github.com/Geektoolkit/Dynaframe3/releases

Geektoolkit, 2020-12-09:
This file is giving it a problem: 

/home/pi/Pictures/BVI_Matt_Julie/._2006-05-16-159.JPG

When it hits that file it crashes and restarts. It sees the full list, just regenerating when it crashes. Can you try removing that image and see how it does? Also can you try to open that file and see if it's corrupt at all?

Log section below: 
12/8/2020 6:38 PM:Beginning to load next file: /home/pi/Pictures/BVI_Matt_Julie/._2006-05-16-159.JPG
12/8/2020 6:38 PM:ERROR: Exception: System.ArgumentException: Unable to load bitmap from provided data
   at Avalonia.Skia.ImmutableBitmap..ctor(Stream stream)
   at Avalonia.Skia.PlatformRenderInterface.LoadBitmap(String fileName)
   at Avalonia.Media.Imaging.Bitmap..ctor(String fileName)
   at Dynaframe3.MainWindow.

driftinaway, 2020-12-10:
[Dynaframe_FileNames.xlsx](https://github.com/Geektoolkit/Dynaframe3/files/5669828/Dynaframe_FileNames.xlsx)

I tried to narrow this down some today.
1. Updated to 2.13 - no change
2. The playlist dumped to to the log is duplicating files and adding the ._ in front of the filename.   for the above mentioned file - the real filename is 2006-05-159.JPG    which opens fine by itself (same for other files it crashed on)
3. I removed the file - no change just crashed on a different file with ._ added to filename
4. It does not always crash at the same file, but always one with the ._ added to the front
5. I copied 3 of the files it crashes on to a new folder with 20 other pictures and ran just that playlist - no issues and no ._ added to the front of any file
6. The same issue presents in another folder with 100+ images
7. I though it might be file size related - so I dumped the filenames and the playlist from the log file to excel, matched everything up and sorted by file size - but does not seem to track with large files.  Spreadsheet attached. Filenames in red are the duplicates in the log file playlist where ._ was added. 
8. I saw mention of mouse and VNC mentioned elsewhere so just for information this is all running with only a monitor connected and no VNC connection established

Does the ._ denote a temp or cache file that's being generated then picked up and added to the playlist causing problems?

qwksilver, 2020-12-10:
taking one example:  what is the actual full path and filename for:
`/home/pi/Pictures/BVI_Matt_Julie/._2006-05-16-159.JPG`

is it /home/pi/Pictures/BVI_Matt_Julie/2006-05-16-159.JPG  and the ._ are added erroneously, or is something else being replaced. 

also the playlist dump labels out a series of files that have the ._ added in, are they always the same files.
if so, rename a few, even if you manually delete the first character and anything "hidden" before it, and put the name otherwise to what it is now.  does it vanish from the list,

connected to that, have you noticed that some are direct from camera and others have been croped/editted/resaved.

Geektoolkit, 2020-12-10:
Thankyou for trying the upgrade. That helps me out greatly by having me work on one codebase so I can try to reproduce and figure out the issue.

The renaming is a bit baffling, I don't manipulate the files in any way (on purpose..I'd hate to do anything to someone's files that may destroy them, and there's no reason to manipulate them).  I don't do any caching, renaming, or file writing at all that I'm aware. of.  I just looked through the code and didn't see anything.

That being said, my code shouldn't crash and not be able to recover gracefully.  I am going to revisit the logic of restarting the playlist on a crash and try to get the next image instead.   I'll also rename some files manually and try to see what's going on there.  Finally I'm going to look for any code that may touch a file, but I'm really not aware of any.  I'm not seeing this on my frames.  

What's the OS..is this a clean install of raspberry pi os? I'm wondering if anything else could be in the environment touching the files.

Geektoolkit, 2020-12-11:
Still looking into this one. Not seeing anything on renaming in the three frames I have setup. Going to try duplicating some of the names given to see if there is any odd parsing.

driftinaway, 2020-12-12:
To answer your questions from a few days ago  
"taking one example: what is the actual full path and filename for:
/home/pi/Pictures/BVI_Matt_Julie/._2006-05-16-159.JPG"

The actual file path is
/home/pi/Pictures/BVI_Matt_Julie/2006-05-16-159.JPG

Note: the playlist always has both the correct filename and the one with ._ added.
The sort order doesnt make sense to me either but I didnt really care so have not really examined further. 

A few spot checks from the logs across 2 days and it looks like its adding the ._ in the playlist to the same files.

Yes its a clean install of the OS - just did updates, VLC and Chromium
I will work some more re-naming and troubleshooting to try and narrow it down further.

Thanks!

driftinaway, 2020-12-12:
[LS_list.txt](https://github.com/Geektoolkit/Dynaframe3/files/5683825/LS_list.txt)
[findlist.txt](https://github.com/Geektoolkit/Dynaframe3/files/5683826/findlist.txt)

Ok it looks like the ._ files are actually there but not shown with ls command or via file explorer on the GUI    
I did a **find** command to compare how may files I had in each folder and the find command lists the ._ files

Some quick digging and its a function from different OS (Mainly Apple) which makes sense as my wife used a macbook and sorted allot of our pictures up until last year.
https://apple.stackexchange.com/questions/14980/why-are-dot-underscore-files-created-and-how-can-i-avoid-them

I am hopping with this info you can account for them as my end goal is to mount folders from my NAS RAID drive and randomly display pictures from the thousands I have there. But I am sure ._ files are scattered throughout if this sample is any indication.

driftinaway, 2020-12-12:
I ran the following command on the folder with the images and so far its running much smother and showing images in an order that makes sense. (Previous order was all over the place)

`find . -type f -name '._*' -delete`

Geektoolkit, 2020-12-13:
SWEET Ok, that helps greatly thankyou...so if I'm understanding correctly, the files are coming from extended info that MAC/UNIS filesystems are storing.  But Dynaframe isn't creating them.

However when Dynaframe goes to play them it pukes.  If they're extended info I'm imagining they're not actual image files, so this is making everything make sense.

I believe the fix from Dynaframe's side is to ignore anything starting with " ._ ".  I can't see many people purposely naming an image file that, and it should be a quick, easy, and safe fix. I'll work to get that into the next release. I have some shuffle bug I also need to fix, so maybe in the next day or so I'll get a prerelease up for you to test. Thankyou for digging in, that helped immensely!

Geektoolkit, 2020-12-17:
Fix coming in 2.14.

closed 2020-12-17 by Geektoolkit