Geektoolkit/Dynaframe3 #50: Problem with using network shares. Looking for some guidance.

Cydoemus, 2020-12-09:
I have set up the fstab to have persistent access to a windows network share. I have also set the raspberry pi config setting to wait for the network on boot. 

Dynaframe will startup but then crash when trying to transition to another photo. 

See attached Dynaframe log. I feel there is a problem with the image list being referenced on run.
I'm not sure what else I can alter to prevent the application from crashing.

[dynaframe.log](https://github.com/Geektoolkit/Dynaframe3/files/5664088/dynaframe.log)

Geektoolkit, 2020-12-09:
Ok I'm taking a look.  Some questions to help me out:
Do you know how to use 'top' to see if the process is still running? 
Does it show any images at all, or fail immediately?
Is there another folder chose that isn't a network folder?
Is /home/pi/Photos/Asa the network share?
If you launch it manually (From /home/pi/Dynaframe launch "Dynaframe") doest it appear to work?
Which version is this? If it's not 2.13 can you please try that one, as it's got some playlist updates.
Are you VNC'd into the device, or have a mouse hooked up? Does it repro if you aren't VNC/d in or have a mouse hooked up? (We have reports that may be causing issues).

From the log it looks like it sees 9 items, and then 18 items on the next refresh which is kind of how I have it designed..it should go through what it can find, and once that's done it rechecks the folders to see if new files are available.  

So I just have to figure out what is it about the scenario that's causing the failure.

Cydoemus, 2020-12-10:
The Dynaframe process is still running.
The first image is shown then it appears to crash at the start of the frist transition. 
the only folder currently selected is the mounted network share.
The network share is /home/pi/Photos/
Launching it manually after killing the tasks has it working for a longer period of time. Then it stops again.
Currently running Version 2.13
I have confirmed that the issue occurs regardless if VNC is in use.

Geektoolkit, 2020-12-10:
ok. I've seen this a few times. Typically it's always on a mouse move, which makes me think unclutter is doing something bad.
Do you have a mouse connected? If so can you try to reboot without it. If not then this may be something else.

The fact that the process is running tells me it's the UI thread crashing, which may mean some kind of rendering issue. If the whole thing went down I'd suspect something to do with file access.  Either way I'll work to repro it on my side in the meantime in as close to yoru scneario as I can get to.

Cydoemus, 2020-12-10:
I did have a mouse connected. I have removed it and rebooted and everything appears to be working correctly. Thanks for the assistance.

Geektoolkit, 2020-12-11:
Yay! Ok I'll look into an alternative to unclutter or see if that may be causing it

Geektoolkit, 2020-12-11:
Closing issue for this one.

closed 2020-12-11 by Geektoolkit