Geektoolkit/Dynaframe3 #51: After some time no longer in shuffle mode

Archonw, 2020-12-12:
Hi,

first, gib thank you for this great work. 
 
In version 2.13 after some time the order is no longer in shuffle mode. I have to set it via the web ui again.
At this time i use a wabdav mount point with over 6000 pictures to feed the frame.  Everything else is working fine for me.

Geektoolkit, 2020-12-13:
Ok. I am tracking some issues where 'shuffle=off' seems to almost be getting set by default when accessing the web UI.  When you check the webUI are you seeing the setting as 'off' or is it showing as 'on' but not shuffling? That'll help me track it down.

qwksilver, 2020-12-13:
setting is showing as on
but i can confirm my file order is quite obviously sequential alphabetical.
i also froze everthing by shuffle off, intending to turn it back on again.

qwksilver, 2020-12-13:
[dynaframe.log](https://github.com/Geektoolkit/Dynaframe3/files/5684282/dynaframe.log)

some interesting findings, comments in the file.

a. i suck at code, but can sometimes follow along
it looks loopy when seeing the log output, and it is writing the playlist out multiple times.

so can it grab the playlist in name order and list that to the log
log that it's shuffling, or log no shuffle because "false", so we can see that it's actually doing so as explected
and if appropriate then run the shuffle module 1 time, and list the result as the new playlist (or log the selector array discussed below) 
then use that as the definitive playlist until the next reorder (aka files changed in the playlist folders)

there is also some discussion in the linked thread lower down about using a cryptorng vs systemrng to improve on randomness.  near [here](https://stackoverflow.com/a/32666693).  and some discussions about using an array of numbers 0 to totalfiles-1.  I think loading that array to a second array and using it as the sequential pick order, with the ability to pick up new files, and mix them in without killing the base list.

and mostly being way over my head, but i take away this, if you use a numbered list to define the playlist position to play next, then can we log that selector array result, or the "shuffled" list based on that result.

Geektoolkit, 2020-12-17:
I'm going to attempt to make this better in 2.14, with a fix further on for better shuffling. The shuffle method doesn't seem to be terribly effective. I've done work to try to improve this on all fronts, but will take some testing to see if I've been successful. I will work on this and get it better for both sorting and for shuffling

Geektoolkit, 2020-12-22:
This should be fixed in 2.15. Please reopen if not. I did signifigant work in 2.14 and 2.15 to improve this.

closed 2020-12-22 by Geektoolkit