Geektoolkit/Dynaframe3 #52: Idea: Settings Menu Revamp

qwksilver, 2020-12-13:
Extended part of this idea is to have the save button, or all the buttons on the control screen, return you to the screen (fully expanded) that you left from without having to click the tab bar again.

The root of this is the radio buttons, replace them with the blue buttons, clicking on it creates and instant change, and returns you to the page you are on like you didn't leave.  Eventually the buttons might pick up a color like green on the on button, and red on the off button, depending on the current setting with the other button always in blue.

useful for all things with radio buttons.
definitely needed for shuffle, audio for video.  (these two could probably go to controls, which would make room for other options in the file sorting module)

the key behavior that seems kinda klunky is not going back to the control page when you hit a button there, or not returning to the settings/playlists page when you "save".

RichN001, 2020-12-17:
I'll likely address some of this on my interface revamp. 

i'll post here when i get something to present.

RichN001, 2021-01-16:
if anyone would like to test some of this WebUI that would be great. I do not have multiple frames, Sync or multiple playlist setup. 

I still have to clean the code up and fix a few CSS classes.. but it should be fully functional. It scales to mobile devices much better. I also made it return to the tab you where on after save. 

Id like to re-organize some things as well as toggle switches for on off functions but this is bit more complicated and I wanted to get a clean responsive UI out and improve some current functionality first. 

You will have to sftp in to your current Dynaframe web directory. Create a new folder called old, and extract the contents of the attached ZIP.

Your Web Directory should be structured like this. 

Dynaframe->Web->css
Dynaframe->Web->images
Dynaframe->Web->js
Dynaframe->Web->WebTemplate.htm
Dynaframe->Web->Old

To go back just delete everything from old and put the contents of old back in the Web dir.

RichN001, 2021-01-16:
doesn't look like I can attache a file. here isa link to download the web dir. 

[https://xgraphics.com/dynaframeui/web01.zip](Link)

RichN001, 2021-01-16:
Some possible movement of items. 

**Settings**
Directory to add to Playlists
Shuffle to Controls
Play Audio with Video to Controls

**Change Button color or make a toggle button for controls of active items**
Turn Screen Off / On - This would have to be included as a variable to pass back. ex 
Show Date & Time - This would have to be included as a variable to pass back. ex 
Show Filename - This would have to be included as a variable to pass back. ex 
Show IP - This would have to be included as a variable to pass back. ex 
Hide Infobar - This would have to be included as a variable to pass back. ex 
Shuffle  On / Off
Audio On / Off

Dimming the page with a Saving Notification might be useful too to Physically giving the user notification things are saving.

Geektoolkit, 2021-01-19:
Thanks Rich. I'm having a busy time at work, hoping to get back to this next week and give this a shot. I'll try for sooner if i can, including the suggestions, those make sense and are ones I'd planned on doing.

qwksilver, 2021-01-20:
I'm also away but will be able to test it on a couple connected frames this weekend

qwksilver, 2021-01-25:
@RichN001 This looks Great!
only issue i ran into was on the playlist screen, it's cramming all the folders and checkboxes together in one big long line, there is no "newline" before the next folder.  I also like the list of Settings, to move these things Directory to add to Playlists, Shuffle to Controls, Play Audio with Video to Controls.  I love that clicking a menu returns you to the menu you are in.  It would be nice if the control screens didn't need to reload that particular page in most buttons.

RichN001, 2021-01-25:
Glad you got to try it.. ill look in to things.. 

I also found the show file name control was not working.. I fixed that. 
Show IP not working either but that might not be me, I dont think it was working for me on the last versions. Ill check.

RichN001, 2021-01-26:
Playlists, looks like I got some work to do on jsLists.js I have not really looked in to that, so I will dig in on that one and see how its being implemented fully. Possibly I could correct the css to work.. just not sure yet till I dig in. 

Ill look in to not refreshing page from controls, I think I can do that in Javascript... Ill play with that. 

I need to do a fresh install on one of my test pi boxes, But my infobar seems wacky, sometimes it just says "Loading Images..." and sometimes it works correctly, although I can't seem to get IP to load. Ill see what happens on a fresh install.. make sure I didn't mess something up on all the changes I've made to it.

qwksilver, 2021-01-26:
fairly simply since you only need delete the dynaframe directory. 
i'm getting clock ok, no filename or ip, hide infobar will rehide clock

RichN001, 2021-01-26:
I fixed the Show Filename button, the zip on the link above was updated for that if you wanted a new copy. 

No IP on a fresh install... COMMAND=INFOBAR_IP does not seem to be working Ill take a look and see if I see what's going on.

qwksilver, 2021-01-26:
out of curiousity, why not fork and generate a PR for the new screens.....

RichN001, 2021-01-26:
just a backup nothing needed in there. Ill add it to my fork and release it. I just wanted to get feedback on things before I made it available with out committing to it I guess.. haha. 

I also created a FLASK based upload and directory management which im excited about that would run on another web port. I want to work on implementing the web functions in that. I probably could take care of the refresh page issue with controls too... so it would just run in tandem to the current webUI at this point.. but possibly merge the 2 at some point and just use FLASK. 

So saying that out loud, I may just wait to fork out the flask version if it works out.

qwksilver, 2021-01-27:
i'm a big proponent of sooner beats later, having your own dev fork and just pushing over desired changes makes so much sense, anyway downloading and installing the updates tonight.

RichN001, 2021-01-27:
I pushed the changes on my fork... Once I get the jlist issue fixed and move some controls around, and see if @Geektoolkit has anything to add Ill do a pull request. It would be a pretty good clean up of function for a release.

qwksilver, 2021-01-27:
It looks great, i might request a dark mode switch down the road, most sites are getting away from bright background tones.

RichN001, 2021-01-29:
Can add that no problem. It would be nice too have.

RichN001, 2021-01-31:
Cleaned up a few more things... multiple fields, tab 3 was not returning...
I need to work on that playlist rendering next. Hopefully Ill get time to get that sorted in the next day or so, then Ill move a few things and send this off for next release.

RichN001, 2021-01-31:
you could just grab the html from my fork, I think thats all I've changed. maybe grab the web directory..lol

qwksilver, 2021-02-06:
this whole discussion is forked in 2.16 there are a few minor things still but i will open a new issue to clean up the settings discussion

closed 2021-02-06 by qwksilver