Geektoolkit/Dynaframe3 #53: Kioskmode for Dynaframe3

masterElmar, 2020-12-15:
It would be interesting to improve the boot up and starting process.
I would suggest that matchbox could open the omxplayer on raspberry pi without using the lightdm.
But also adding the whole startup into the rc.local file could be a huge improvement in startup for a clean
"digital signage" or picture frame. Because your project is amazing, i love it <3.

With kind regards
Elmar

Geektoolkit, 2020-12-16:
I'm looking into this. My concern is that it looks like matchbox is a single window at a time. We actually currently use 2 windows for Dynaframe (Dyanframe is an AvalonUI executable, and then we spawn omxplayer).  

The boot time should be a one time cost, so for commercial installations / digital signage once it's launched it should be done. So while this may help with first run it'd limit future features and capabilities that I want to retain for flexability.  For instance I have a goal of having it launch other apps besides oxmplayer in the future (visualizations, web browser, even mame possibly).  I'm going to resolve this as won't fix for now unless the startup time is demonstrated as something that is blocking for some reason. I appreciate the suggestion!

I'm going to keep this open though as I investigate rc.local.  I believe this was giving me grief because I needed to show UI, however I'll revisit it.

RichN001, 2020-12-17:
It would be nice to get to that point maybe over time something will come about. until then you could hide the task bar and set the desktop to black. 

you'll see the task bar for a second. (Actually I think there is a way to disable the task bar) so you could eliminate that brief sight of it. 

adding a logo on a black background might look nice or i'm sure there is a way to do a gif animated background. 

just a workaround, certainly wouldn't speed anything up but would look more together to guess haha.

RichN001, 2020-12-21:
Just for the sake of putting it out here for anyone that wants to do this and to save them from looking it up this is what I did, here are my notes of things changed to go (appear to go) straight to Dynaframe as possible. Ill attach a video as well. 

You get some on / off of the backlight and a small very fast group of text at the end, It has not bothered me enough to see what it is or try to stop it yet haha.. If anyone comes up with something better or different please post. Ill also update any new findings if I come back to this. 

[Video of boot](https://hiimpactstudio.net/boot/IMG_0363_boot.mov)

**_Hide Trash and set Desktop background_**

sudo nano /home/pi/.config/pcmanfm/LXDE-pi/desktop-items-0.conf

Change show_trash=1 to show_trash=0

(I also disabled show_mounts and set background to black no image, my file looks like this for black screen)

[*]
desktop_bg=#000000000000
desktop_shadow=#000000000000
desktop_fg=#e8e8e8e8e8e8
desktop_font=PibotoLt 12
wallpaper=
wallpaper_mode=color
show_documents=0
show_trash=0
show_mounts=0

(With Dynaframe logo background, my file looks like this)

[*]
desktop_bg=#000000000000
desktop_shadow=#000000000000
desktop_fg=#e8e8e8e8e8e8
desktop_font=PibotoLt 12
wallpaper=/home/pi/Documents/pibackDF.jpg
wallpaper_mode=3
show_documents=0
show_trash=0
show_mounts=0

CTRL + O to save 
CTRL + X to exit

Sudo reboot, task bar should be gone (reverse this to get it back).

**_Remove Task Bar_**

Edit Auto Start File

sudo nano /etc/xdg/lxsession/LXDE-pi/autostart

comment out the line @lxpanel --profile LXDE-pi with # symbol

CTRL + O to save 
CTRL + X to exit

Sudo reboot, task bar should be gone (reverse this to get it back).

**_Doing away with some boot stuff_** 

I added the below in bold to my boot cmdline.txt. You get a few flickers in black and then your desktop. 

sudo nano /boot/cmdline.txt

console=serial0,115200 console=tty3 root=PARTUUID=6c4f6a69-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait **consoleblank=1 logo.nologo quiet loglevel=3 plymouth.enable=0 vt.global_cursor_default=0 plymouth.ignore-serial-consoles splash fastboot noatime nodiratime noram**

CTRL + O to save 
CTRL + X to exit

Sudo reboot, task bar should be gone (reverse this to get it back).

qwksilver, 2020-12-28:
it occurs to me that it maybe possible to just unload the desktop completely, @Geektoolkit  does dynaframe actually require it to function, and if it's not there what would the bootup looklike?

qwksilver, 2021-02-06:
i will let you know now that unloading the whole desktop was not workable.

qwksilver, 2021-02-06:
@RichN001 is this https://github.com/Geektoolkit/Dynaframe3/issues/53#issuecomment-749205299 including in 2.16 also, or is there still more to do to streamline startup, or have we hit the pi limits.

RichN001, 2021-02-06:
i think streamlining is still on the table.

qwksilver, 2021-05-28:
This issue is as finished as it's likely to get, if others have ideas specific to making boot faster or darker please open a new issue

closed 2021-05-28 by qwksilver