Geektoolkit/Dynaframe3 #55: Mixed Videos and Images Bugs

ayushsaklani, 2020-12-18:
Hi, 

I really like this project and thanks for making this. 

In version 2.13 I faced few problems that might be categorized as bugs. I using raspberry pi 3b+ and latest Raspbian OS. 
I have mixed folder for images and videos. 
1.  If we exit Dynaframe with ESC button. Then still videos are playing but images are not. So, in playlist after few seconds video is playing and coming back to desktop and again after some time video file is playing and back to desktop again. 
2. Transition from image to video is smooth. But after the video playback loading of next image file take time and screen is blank for that time period(around 2-3sec). I have set my transition time to 0.

qwksilver, 2020-12-20:
1. can you let us know if you are on vlc or mouse/keyboard
2. if the video is shorter than the slide time, then there will be a blank screen while waiting for the next slide to fade in, fade and fade out don't impact video at this time.

ayushsaklani, 2020-12-20:
1. Raspberry pi is connected to my TV. I use vnc/ssh server sometime to connect from phone if i have to do some changes on Raspberry pi. 
2. The video length is 30 secs and i have kept slideshow timing to be 15000ms (15sec). I tried keeping 10sec also but still same issue.

Geektoolkit, 2020-12-21:
I'll take a look at these when I get my dev setup going. Thanks for getting it going and reporting the issues. One other ask is can you exit via the web UI and let me know if you hit the same issue with the videos continuing to play. Exiting is a little sloppy due to an issue with .net core currently.

ayushsaklani, 2020-12-22:
Hi,
After exiting from webUI. Intermittent video playing issue is not happening.

Geektoolkit, 2020-12-22:
Ok. That may be a known issue for the intermittent part. If you exit it doesn't exit cleanly (this is due to a .netcore bug I can't fix until I move to 5.0 from 3.1).  That's currently 'external' and not soemthing I can fix. I expect if someone exits dynaframe they'd either sudo pkill Dynaframe to clean up, or reboot to cleanly restart until I can get a fix in there.

I'll have to do more testing with a '0' transition time (I believe we have a built in 1.5 second delay expecting a transition, and to allow the video player to exit after we send the kill command).  I'll test a bit with that and try to see if I can get a repro.

qwksilver, 2020-12-28:
ok, a bad exit of player, especially loging in with vlc, stops image playback but not stopping dynaframe itself.  so anytime a video pops up in the playlist, it runs the video on the screen, but not through the player.

closed 2020-12-28 by qwksilver