Geektoolkit/Dynaframe3 #56: VLC Player Discussion

qwksilver, 2020-12-18:
ok on pc so far this command will run the first 14 seconds of a video, and then quit
vlc --run-time 14 --stop-time=14 vlc://quit <videofilename>

i know there are commands for orientation of player, video, fullscreen no borders, and zoom modes i'll tinker more in a bit

qwksilver, 2020-12-18:
reference locations: https://wiki.videolan.org/VLC_command-line_help/

vlc <filename> --start-time=14 --stop-time=28 --fullscreen --video-filter "rotate{angle=0}" vlc://quit

this has 2/3 of the desired functions and 1 bonus of starting outside frame 1, however the video will stutter at the beginning if it's not exactly on an i frame.

Geektoolkit, 2020-12-19:
If you get a chance can you let me know how performance is of VLC on a pi3/4. I may try to switch over to it. I also am thinking of having an advanced tab with "additional command line options" and then that'd unleash all of the command line power for someone to use. VLC could also enable stuff like being able to view security cameras (RTSP) and widen the file formats greatly. 

My main setup will be down for a few days while my office is rebuilt. Hoping to get back online as soon as I can

Geektoolkit, 2020-12-22:
I see this discussion on it: https://github.com/raspberrypi/linux/issues/3095.  When using the UI, right when I click 'transform" I get a blank screen. Using "cvlc Terminator.mov --stop-time=14 --fullscreen :vout-filter=transform --transform-type=270 --video-filter "transform{true}" I get the MMAL errors (This is on a pi4) 

_[b26c6f18] mmal_codec filter error: Failed to enable output port vc.ril.hvs:out:0(OPQV) (status=3 EINVAL)
mmal: mmal_vc_port_enable: failed to enable port vc.ril.hvs:out:0(OPQV): EINVAL
mmal: mmal_port_enable: failed to enable port vc.ril.hvs:out:0(OPQV)(0xb26a7430) (EINVAL)
[b26c6f18] mmal_codec filter error: Failed to enable output port vc.ril.hvs:out:0(OPQV) (status=3 EINVAL)
mmal: mmal_vc_port_enable: failed to enable port vc.ril.hvs:out:0(OPQV): EINVAL
mmal: mmal_port_enable: failed to enable port vc.ril.hvs:out:0(OPQV)(0xb26a7430) (EINVAL)_

EINVAL looks idfferent than ECORRUPT and ENOMEM so I think we're hitting differnet issues. But...we know it's not a simple command line swap now, which while dissapointing, is at least info we can use.  This and my experience with VP8 makes me question if we should keep fighting this path, or just say 'convert your webm over to mp4 for this".  I even thought about just including a converter and trying to do it in the background...so if it saw a webm it would create a proxy. but once Dynaframe starts writing files, things can get hairy (Say someone has a NAS  full of WebM and all of the sudden DF is trying to keep up and is writing files..somewhere...(back ot the NAS? to an internal storate on the SD card? what if it fills it? etc)

Just stuff to think about. I've played full screen gifs with a different app that we can look at to support those...I'm not sure how popular 1080p gifs are though.

qwksilver, 2020-12-28:
Nothing either of us can find indicates that VLC is currently viable, I think you mentioned an open ticket to expand support in omxplayer.

Tabling this issue to work on more important things, I'll reopen if I find any better information.

closed 2020-12-28 by qwksilver