Geektoolkit/Dynaframe3 #61: Random issue

qwksilver, 2020-12-28:
1st a question, my subframe is reporting no shuffle even though shuffle is on in settings.  the file list is sequential by filename, not directory based, and says list generated 53 items, shuffle is off.  is there logic that tells it if it's receiving the files from somewhere else, then don't shuffle file listing in this machine. 

on main machine 
-lists files 
-says eb1.jpg is file 1 
-opening file eb1.jpg
-decidest to reshuffle and displays new list
-note new list is way different from first list, and new first file is 11001001.jpg
-says opening eb1.jpg

/smh(wall)

Geektoolkit, 2021-01-04:
Next chance I get I'll work on adding 'live logging'.  That should help a bit.
The logic is 'get all of the files from all of the directories', then 'shuffle resulting file list' and then it should start going through it. If a file doesn't load it may reshuffle (I think I changed that behavior though). Also any change to the UI that may change the file list will reshuffle it. 

This is one of the areas I want to work on before next full release. I also plan on adding in the sort by date and such in there, and since that will all touch the same code I may do that first, and try to fix shuffle in that update.

Geektoolkit, 2021-01-12:
oh, I just realized I didn't answer the subframe question. There is no logic or communication of settings across frames.  The way it's designed currently is that the main frame can be told to tell other frames what to do.  Then those frames should be 'paused' if full control should be given to the main frame, or they can be given whatever timeouts someone wants...but if the main frame synces them, the timers will conflict.

Ideally if you have multiple frames, you tell the main frame 'control all of these' and then pause everything else.  Then things should work swimmingly.  If the main frame is set to shuffle, then it'll generate a shuffled playlist, pick a file, and tell the other frames which file it chose (they then try to get as close to it as possible, using folder mathcing, file name matching, etc.)

My thinking was to minimize cross frame communication and management. please try the experience where the subframe is paused and let me know how that looks to you. For my 3 frames, there are slight sync issues here and there (I expect that since one is a pi3) but overall the effect feels 'clean'.  However that's with images, I have a feeling when we get to video/cinemagraphs that'll change drastically.

qwksilver, 2021-02-06:
so yes pause works, however it's tedius, and I am noticing a few minor issues all connected to this system.  

1. it doesn't handle a missing directory, possibly blocked directory, well, ie no chance to match means I get the same default photo over and over while paused. 
2. timing is still a lot shit, synchronization of matching video will require a present filename and a very detailed start time based on system clocks.  NTP can keep the various frames time identical.
3. i recomend the addition of an auto-pause function on sub-frames  perhaps change the on-off buttons for "Main - Off - Receiver' on the sync page.  if receiver is on then start frame paused, there is also room for a backup playlist if time since last timecode is > 25% more than the slide tiem then backup plays.
4. also give consideration to the desire to pause a syncd pair/set for a long period of time, haveing main send out "longpause" to all subs would stop the whole set.

qwksilver, 2021-05-28:
randomization appears to be more random, more work later maybe, but it's mostly good now.

closed 2021-05-28 by qwksilver