Geektoolkit/Dynaframe3 #67: Error accessing web controls when directories have been removed

qwksilver, 2021-01-11:
Fatal Error! Excpetion occurred.  Info: System.IO.DirectoryNotFoundException: Could not find a part of the path '/home/pi/Pictures/Testport'.
   at System.IO.Enumeration.FileSystemEnumerator`1.CreateDirectoryHandle(String path, Boolean ignoreNotFound)
   at System.IO.Enumeration.FileSystemEnumerator`1.Init()
   at System.IO.Enumeration.FileSystemEnumerator`1..ctor(String directory, Boolean isNormalized, EnumerationOptions options)
   at System.IO.Enumeration.FileSystemEnumerable`1.DelegateEnumerator..ctor(FileSystemEnumerable`1 enumerable, Boolean isNormalized)
   at System.IO.Enumeration.FileSystemEnumerable`1..ctor(String directory, FindTransform transform, EnumerationOptions options, Boolean isNormalized)
   at System.IO.Enumeration.FileSystemEnumerable`1..ctor(String directory, FindTransform transform, EnumerationOptions options)
   at System.IO.Enumeration.FileSystemEnumerableFactory.UserDirectories(String directory, String expression, EnumerationOptions options)
   at System.IO.Directory.InternalEnumeratePaths(String path, String searchPattern, SearchTarget searchTarget, EnumerationOptions options)
   at System.IO.Directory.GetDirectories(String path, String searchPattern, EnumerationOptions enumerationOptions)
   at System.IO.Directory.GetDirectories(String path)
   at SimpleHTTPServer.GetDefaultPage() in C:\Users\joefa\source\repos\Dynaframe3\SimpleHTTPServer.cs:line 464

qwksilver, 2021-01-11:
the above is what i get when opening the web url from my main computer

Geektoolkit, 2021-01-12:
ok i see where that is, I'll work on it for the next bugfix update.  This makes sense where it's expecting things to be there that are in the settings file, and it's not getting a chance to clean it up. I'll have to add some tighting around how the setting errors are handled.

qwksilver, 2021-02-06:
apparenly in the new ui, or a fix just prior to that, no path on boot reinstalls the default paths automatically.  this error is no longer an issue

closed 2021-02-06 by qwksilver