Geektoolkit/Dynaframe3 #70: Feature enhancements and questions

McDaver, 2021-01-17:
Questions:
1. If I move the mouse when on the raspberry pi the Dynaframe stops displaying pictures, which is fine, but how do I start it again? Maybe you could add this option to your control screen.
2. If I want to display pictures from my NAS, how do I do that?  I've tried putting in a smb://[servername].local/photo for example and that does not work.  I opened the NAS shared folder directly on the Raspberry Pi and navigated to the photo folder.  I then used that path which looks like /photo on syn918plus.local/ but that doesn't seem to work either.
3. How do I get search directories to show up?  Mine is blank under the Playlist menu.  

**Enhancement thoughts**
1. I would move the ability to add folders under your Playlist menu rather than in Settings and maybe call it file locations and playlists, or something like that.  It should include the ability to provide credentials so you can just access the files directly from cloud servers or NAS devices that require credentials.  I could work with the mapping option on the Raspberry Pi but I have been unable to get that to work has mentioned above.  Is using a standard UNC path supported?
2. When hitting Save button on the settings menu, there is no feedback that it actually saved.  The changes to the aspect ratio and other settings that have "Current Value" displayed, don't change after hitting save, so it looks like nothing happened.
3. Have an option to set different picture folders/locations to different displays. One can use an adapter and hookup additional displays to the usb ports.  I know Displaylink drivers are available for the Pi, see here: https://support.displaylink.com/knowledgebase/articles/1883065-using-arm-binaries-on-raspbian-on-raspberry-pi (maybe due to driving multiple displays a Pi 4 is required, I do not know.)  Once can buy a usb to HDMI, DVI or VGA adapter for about $15.  Could then drive another display from a single Pi.  I'm going to try hooking up another display to my Pi using such an adapter and see how it goes.

skiper1a, 2021-01-19:
some problem here:

If I move the mouse when on the raspberry pi the Dynaframe stops displaying pictures, which is fine, but how do I start it again? Maybe you could add this option to your control screen.

Geektoolkit, 2021-01-19:
This is due to 'unclutter' which is hiding the mouse. It conflicts with the UI, so if you move the mouse it crashes the UI thread.  The main way to get out of this state is to launch a console and type "Sudo pskill dynaframe"  (may have to capitalize it, I have to check when I get home).  I'm working on a better solution, but for now the gudiance is when it's showing images, unplug the mouse unless you're managing it.  It's unfortunate, but until I get a better way to hide the mouse I'm stuck with it.

The reason videos show is I 'shell out' to a video viewer for those, so omxplayer works.  Skpier1a is correct...the process is still running, it's just the rendering that gets torn down (The UI thread is actually still functioning...it's been maddening to debug)

For displaying from a NAS, you'd mount the folder locally.  I don't have UNC support currently...I'll look into it. If you mount the folder then it should 'just work' like any other local folder.

I'd expect search directories to be there...which version was this on? I'll check to see if I regressed something.

qwksilver, 2021-01-20:
replacing run.sh entirely with this code, might fix the mouse issue.  it will also add a new log file for each reboot, and cleanup log files after 10 days.  if you want to check that out and let us know that it does (or doesn't) solve the mouse issue that would also be helpful.

`#!/bin/bash

# This file gets run to launch Dynaframe from the autostart command on Linux/Raspberry pi systems
# This is needed to get the 'working directory' synced up. I'm keeping it because I realize it can be used
# to also shim in other fixes without affecting the main codebase which are linux specific before execution, such as turning off
# sleep, or possibly syncing files.
# unclutter is installed by install.sh and used to hide the mouse cursor
# setterm is used to prevent screen blanking
#
# filename/date article on bin/bash/scripting https://www.cyberciti.biz/faq/unix-linux-appleosx-bsd-shell-appending-date-to-filename/
# delete older then 10days on stack overflow https://stackoverflow.com/questions/13489398/delete-files-older-than-10-days-using-shell-script-in-unix

# unclutter -idle 2 & (redundant command that is running elsewhere)
cd /home/pi/Dynaframe
mkdir -p  /home/pi/Dynaframe/logs
echo "starting Dynaframe" >> /home/pi/Dynaframe/logs/run.sh.log

# find and delete any log file older than 10 days, could be longer later.
find /home/pi/Dynaframe/logs/ -mtime +10 -type f -delete

# set date and time, create log file on each reboot
now=$(date +"%Y-%m-%d-%H-%M")
./Dynaframe > /home/pi/Dynaframe/logs/dynaframe-${now}.log 2>&1`

qwksilver, 2021-02-06:
1. mouse issues fixed in 2.16,
2. add investigation of unc support to master list of software enhancements
3. it's likely the save button, type in new directory and click save, it's possile just hitting enter/return doesn't always register, but it does appear to work instantly in the new ui
----
1. playlist dir add moved to playlist screen
2. screen now returns to screen you are on, with refreshed information, thanks to RichN001 for all the ui updates
3. multimonitor support is not yet available through avalonia, which is the limiting factor of multi-monitor distribution from one pi, this will go on the long term wishlist though.

all issues here are closed or slated to go elsewhere, feel free to open new issues if anything isn't resolved completely

closed 2021-02-06 by qwksilver