Geektoolkit/Dynaframe3 #71: Ignore Mac Photo Library

RichN001, 2021-01-26:
We might want to add a bypass for Apple Photos Library "Photos Library.photoslibrary" Ive not built on this Mac yet and I have a large library on here and it is scanning everything (Originals, edits, Resources). Im not sure what its going to do when it hits Lightroom which is even larger haha.

RichN001, 2021-01-29:
Yeah, not that the Pi would have these directory's and Im not sure of Windows defaults, but on Mac in the Pictures folder we could have *.lrlibrary and *.photoslibrary Directories. It would be nice to ignore those.

Geektoolkit, 2021-02-04:
Ok, next version will have a fix for this. I filter out those two by default, but I did it in such a way that someone can edit appsettings to override/add more. I'll consider that an advanced/developer scenario, but it gives us flexability to add it to the UI later so that people can control what does/doesn't get scanned.  I'm avoiding any checkins to the web interface till your PR goes in

qwksilver, 2021-02-05:
ignore folders added in 2.16

closed 2021-02-05 by qwksilver