Geektoolkit/Dynaframe3 #75: Support for more image file formats

wahlstedt, 2021-02-02:
How about adding more file formats? I have a big library of high quality scans in TIFF format that I would like to display without having to convert all the images, but at the moment it seems only JPG, PNG & BMP files are supported.

Geektoolkit, 2021-02-04:
I'm currently limited by what is supported by avalonia. I try to support as many as I can, and will see what I can do here. Tiff has been around a while and hopefully is supported.

Geektoolkit, 2021-02-04:
I just tried this, and Avalonia is not supporting (Currently) TIF.  I also tried TGA since I was in the codebase and got curious. Both resulted in crashes.  Looking further into this, it appears that Avalonia is built on Skia, and skias support is limited by default to png, jpg, and webp.  I'll poke the avalonia forums, but for now the best way is going to use a bulk image converter program. For more info, this is Skia's support, and how skia can convert it, but I don't have access to this level at this time: https://docs.microsoft.com/en-us/xamarin/xamarin-forms/user-interface/graphics/skiasharp/bitmaps/saving#:~:text=As%20you%27ll%20see%20shortly%2C%20only%20three%20of%20these,and%20%28for%20lossy%20formats%29%20an%20integer%20quality%20variable.

Geektoolkit, 2021-02-04:
Closing, will keep an eye on this but for now, I can't fix this.

closed 2021-02-04 by Geektoolkit