Geektoolkit/Dynaframe3 #79: minor error durring ./UpgradeDynaframe.sh 2.16

qwksilver, 2021-02-05:
Restoring old appsettings...
./UpgradeDynaframe.sh: line 79: unexpected EOF while looking for matching `"'
./UpgradeDynaframe.sh: line 82: syntax error: unexpected end of file

Archonw, 2021-02-05:
After installing 2.16 dynaframe won't start and longer. 
Back to 2.15 everything works fine.

qwksilver, 2021-02-05:
2.16 is super alpha, so right now it's got an issue with CR/LF in run.sh  which is why it isn't starting. We should be able to get a repaired file up tonight.

RichN001, 2021-02-05:
the initial install.sh looks to be like the wrong version when you get it form the link. (not in the zip) Its missing a cmd I added to delete a readme file that is a placeholder. But either way, when it copies the correct install.sh to UpgradeDynaframe.sh it seems to be messing it up and then I see the same error you posted above. 

looking at the same file it copies in Dynaframe dir I don't see the error. Im just pasting it in a editor and shows the EOF error. I didn't hunt for anything because it seems like its happening on copy which I scratch my head at, I don't get it. 

I put a build copy on one of my servers with the correct install.sh from the build (altered url to my server) and ran the install command and downloads, installs and runs as expected. 

run.sh runs as it should as well as the copied UploadDynaframe.sh

qwksilver, 2021-02-05:
rich what editor do you run, and are you set for crlf or just lf.
(lf) being correct on linux-arm

also i repaired the run.sh and updated to install package on 2.16 about 10 min after my last post

RichN001, 2021-02-05:
I use Vim, Sublime and CodeRunner on Mac. I just pasted in to Coderunner to see what I could spot... Im at work so I was just pasting around and trying a few things to see it kick.

qwksilver, 2021-02-05:
if your install.sh is differnent drop a pr so we can see the compare, i'm not sure why anything that you just dropped would be missing, but you might want to take a look see, 2.16 is basically just a merge of all the outstanding prs and a few of the commits from geektoolkit that weren't ported to 2.15  the largest problem is that the end of line LF is being converted to windows style CRLF , if you open it on pi in geany and turn on view line ends, you can see it clearly if the file itself is correct or not after install.  as of now if you run from 2.16 everything should be correct, but i will check mine in a bit.

qwksilver, 2021-02-05:
[image]

qwksilver, 2021-02-05:
so i checked mine after reinstall, and my frame is correct.  but i don't know if it's the version in your repo

qwksilver, 2021-02-05:
this is likely fixed, and is fixed for the current distro, closing.

closed 2021-02-05 by qwksilver

RichN001, 2021-02-05:
nice thanks...  what I was saying was if you grab this https://github.com/Geektoolkit/Dynaframe3/releases/download/2.16/install.sh is missing a line so I imagine just the version number got changed and its using a hold one. Its missing my delete line... There is a placeholder readme.txt in uploads. I did that because I needed the folder to go over on build... there may be another way to do that but I didn't see so I just added this file. Then delete it on install.  SO, if you go to uploads and there is a empty image showing in the table its the text file, just delete it. lol

qwksilver, 2021-02-06:
oh i think that maybe that geektoolkit edited his version to add the 2.16 and it overwrote your change.  which is what i think you are also trying to say.

RichN001, 2021-02-06:
i’m trying to say. when there is a release and the Dynaframe2.zip is uploaded also a copy of install.sh is uploaded to the same place.

then the wget link for release points to the install.sh

that copy is not the same.

 https://github.com/Geektoolkit/Dynaframe3/releases/download/2.16/install.sh

click that and look at that file it downloads.. not the same as the release copy. just needs the new copy uploaded to the 2.16 Directory.

qwksilver, 2021-02-06:
ha, sorry i was being so dense, i got it, i merged yours in and then set it to the wget side as well.
its just a file that gets peeled out and added to the installation files.

Geektoolkit, 2021-02-06:
Yeah sorry that was on me. I edited the one in the folder to fix the crlf issue and it ended up not being the updated one. Glad it's fixed up.  I do exactly what you said....I keep it in the zip but peel a copy out to upload to git so that there's one available for the wget commands to hit.  If there's a better way to do it please let me know, I kinda invented that on the fly. I keep it in the source so it can be edited and checked in alongside the rest of the files. I imagine at some point I'll have a 'publish' command that does things like update the versions automatically and such, or come up with a better solution for updating from within the app.