Geektoolkit/Dynaframe3 #83: 2.16 Open & New Issues Sheet

qwksilver, 2021-02-06:
From #59 
- [ ] #8 b : Misc File Formats to still be addressed (WEBM, GIF)
- [ ] #18  : Advanced idea, can it add blurboxing effect to blank space in uniform mode
- [ ] #45 : Add other options for not shuffled playlist sorting, currently only supporting filename
- [ ] #48 : Add COMITUP functionality for AP direct access for initial setup of wifi or direct access to frame outside network
- [ ] #53 : rc.local investigation and improved bootup
- [x] #52 : Settings menu streamlining
New
- [ ] #59 : Live log view

Wishlist
#70 a UNC pathing
#70 b Multimonitor on one PI

qwksilver, 2021-02-06:
#83 e alternate solution is to automatically add the directory on any new upload
might always add it back if the directory isn't empty, but in an unchecked state, as a reminder that space is in use.

#83 f :  so changing ip won't save settings to another screen

#28 video sample of one layout idea: https://www.youtube.com/watch?v=Wz68qwVikV4

Geektoolkit, 2021-02-11:
Rich PR is approved, let me know when you complete it and I'll work on a 2.17 release that we can mark as pre-alpha till we get some time on it, but also letting others play with bleeding edge

RichN001, 2021-02-11:
I think the PR has everything im going to do right now if you wanted to merge it in. Everything else in the UI list Most likely im going to cover in next update from me. Could be a little bit so if you wanted you could do a 2.15.1 or something... If you guys got anything to throw in or are close to it then once your done then go ahead and 2.16 it.

I just wanted to make sure we where not going to delete users uploaded files and fix a few other small things.

Geektoolkit, 2021-02-19:
I'll look to make a release this weekend. I got the sensors to detect rotation to prototype around with that idea, though I think I'm going to do that as a python script outside of this (and I think making a 'scripts' folder with scripts we're using for stuff like that will be handy...I've got a few at this point that may be useful).  I'll work to get into the code this weekend

RichN001, 2021-02-19:
nice, you can grab that motion.py script from discussions too. Although we should work on something to change the delay value and turn on or off in the WebUI. I guess something to turn on/off any of them would be ideal. To them in the release maybe we could do the scripts folder, then have disabled folder in that. Anything outside disabled could run, but by default leave them in the disabled folder. On/off function could move them in and out of the folder. We could probably get scripts to read variables from the appsettings.json im thinking so we could set to run that way or make setting adjustments. then we could have a helper script that could manage settings and if they run or not..  

just thinking out loud.

RichN001, 2021-02-19:
actually if you could hang tight on release, I got more updates, would make sense to wait. I might be a few days thou. im just fixing the url issue and I think I have a solution for the buttons changing colors.

RichN001, 2021-02-19:
actually I should be done tonight. I got a few free hours later so Ill get the PR up tonight its looking like.

RichN001, 2021-02-20:
Im ready, you might want to bring in my PR then Ill do the next one. Im not sure how it will handle 1 on top of the other.. I imagine it would be fine but I dont know.

Geektoolkit, 2021-02-20:
Ok pulled it into the mainline branch. Should I push a 2.17 for testing or do you want to hold off for more changes?

RichN001, 2021-02-20:
yeah hold up i got another update, i’ll be a little bit still but will be done tonight.

RichN001, 2021-02-20:
looks like its in there...

RichN001, 2021-02-20:
ok build away...

Some of the control buttons now toggle, On/Off Green for on... removed unnecessary buttons. Ex. Only needs to be 1 button for Screen command... it switches when you click it. 

I removed Hide infobar because each command button turns it on and off. Id like to rewrite how the string works on the display so that we can stack them.. then you can have time and filename. I know there was some talk on EXIF data so I stopped doing that figuring we would get to it. 

Ive tried a few different methods of getting rid of the url after default.. but every direction I went lead to a issue. So Playlist/Sync and Settings tabs do wipe out the settings so you can't accidentally write them to another frame by changing the ip.. however we still have default.htm?#tab4 so at least it won't change your settings it will just jump to the tab you where on. 

Controls tab still needs to be dealt with... these are href's not submits so I need to come up with another way to do it on that tab. This can go on the Issue list. Oh also I just noticed Pause, that can be a issue added too, pause should be using the new method to toggle on and off... 

Other updates where install.sh to preserve uploads. 

and a few other fixes I mentioned in the PR of small update. 

Im still working towards a new WEBUI but im trying to get toggles with variables, URL rewriting all figured out here first. While still providing some changes that benefit what we got and being able to have us test things to make sure variable additions dont cause a problem somewhere... 

Lets test this thing...

RichN001, 2021-02-27:
Well at least it wast built yet, now at least the Pause button is toggling with the rest in the previous PR. New PR fixes that.

RichN001, 2021-02-27:
My two latest PR's include these items..

#83 a : Move Shuffle to Controls ? - DONE
#83 b : Removing a Playlist doesn't return to playlist screen (returns to controls instead) - DONE
#83 c : Colorize Pause when Paused or change wording to Resume (see d also) - DONE
#83 d : Dedicated Pause/Resume Web Commands - DONE _(CONTROL_PAUSE_On and CONTROL_PAUSE_off)_
#83 e: Add a "Add Uploads Playlist" to screen as button to restore (I made the delete button disabled so that it can not be removed for now)

qwksilver, 2021-02-28:
I updated checkboxes above, but have not yet had a chance to do any testing myself.

Geektoolkit, 2021-03-03:
I pulled in the latest changes as of tonight, created a 2.17 and pushed it out to preview. 
sudo wget https://github.com/Geektoolkit/Dynaframe3/releases/download/2.17/install.sh && bash install.sh; rm -f install.sh 

I DO like shuffle moved to controls, that's definitely nice. And toggle buttons are definitely nicer, though I wonder if anyone will be confused by the behavior.  It makes sense to me after using it a few times, so we can keep our eyes open and see. 

Huge thanks for the work on that release Rich, and as always the testing and support that you both have helped me with. I was busy with life stuff and the project would've not moved if not for both of your help and contributions.

I got the rotation sensors in the mail, after my next yt video I'm going to try that out to see if we can get a script for dynamic rotation going.  I'm excited to try it out!

RichN001, 2021-03-03:
The interface is def evolving. The changes made are functional testing in a sense. for example getting the settings in to make them toggle and trying to move things around the make sense. To me its a bit clunky still with the page jumps, returning to tabs after save. 

Changing all buttons to functional Sliders that show on / off  and other more proper Web App / Mobile App functionality is something im trying to aggressively move towards along with bottom navigation for mobile devices. 

Thanks for getting it in and built up, I know you have been busy. Def excited to hear about the Rotation Saga...

RichN001, 2021-03-03:
found a line we need to add to _install.sh rm -rf /home/pi/Dynaframe/web/ico_ other wise it ask for user interaction. On new installs or even first upgrade with this version it won't come up but anything after this or second install it will. 

Using the install link for 2.17  the version of Dynaframe2.zip you have posted is not running. I didn't compare to my build but it doest try to start even. I didn't see a log dir created either. Ill take a look later unless I see you figured it out. 

Using my last build Dynaframe2.zip works ok there must be something missing in your archive. Below is a link to build on my server for testing. 

sudo wget http://xgraphics.com/dyna//install2.sh && bash install2.sh; rm -f install2.sh; sudo reboot

Geektoolkit, 2021-03-03:
That's wierd I'd tried it last night right before bed (Got too excited had to see it) and I had it working. I'll double check. I did a rm -rf Dynaframe by hand I think before  so maybe that did it. I'll have to test it out more tonight after the kiddos down. Rats.

RichN001, 2021-03-03:
yeah removing Dynaframe directory would not show you the error and also on install from any other version from 2.17 would probably not show the error either as ice directory does not exist until this version. 

I don't know about the rest. I just know I tried with mine last night at the office and it worked fine... Updated my frames when I got home from mine and all was good. 

Today just to test install I ran from GitHub build for the version and Dynaframe not starting, not logs.. so I cleared everything.. tried again and nothing. then I cleared everything and used my test server build and it popped right up, logs good. 

So it just leads me to think the archive from your build is fouled up somehow.. 

if I get time before I leave the office Ill run a compare and see what I see.

RichN001, 2021-03-03:
Tried on another box just now when I got home. I did as you manually rm -rf Dynaframe then ran it install from Github link.. and im on a desktop. No logs in in Dynaframe.

did a rm -rf Dynaframe  again... then ran from my server and it came right up as normal.

Both are same install.sh other then I added rm -rf /home/pi/Dynaframe/web/ico to mine. I even grabbed the latest one and replaced mine at 1 point but changed the URL to my server for download and that was ok too. 

It points to something being up with Dynaframe2.zip.. its weird you got it working thou.. so id say it must be something with how my Pi's are setup but that doesn't really make sense. 

I also tried install 2.16 from GitHub and that worked as expected, so im back to Dynaframe2.zip in the 2.17 build. 

See what you find... Ill be back on later tonight after mine are down.

Geektoolkit, 2021-03-04:
I just did an install and didn't delete the folder, did a sudo reboot, and it appears to have worked. At this point though I'm not sure if I'm in an odd state, I've been installing over the top of itself for ages now.  The version in the web page shows 2.17 (btw, huge thanks for adding that, that looks great with the logo).  I'll have to keep messing with it, i have my other two to try it on to see if they're any different.  

The good news is that when it works it's really clean looking, so probably just a one off thing to figure out.

RichN001, 2021-03-04:
I just did it on another frame I have upstairs and to the desktop I go..and no Dynaframe.. Tried from my server and its back haha... maybe its me.

Archonw, 2021-03-05:
I tried to install 2.17 like i did 2.16, but it dosn't start up. Back to 2.16, no problem.

RichN001, 2021-03-05:
hey.. can you try this build from my server?

sudo wget http://xgraphics.com/dyna//install2.sh && bash install2.sh; rm -f install2.sh; sudo reboot

Archonw, 2021-03-06:
Thank you, this works. No issues so far.

Geektoolkit, 2021-03-11:
Rich I got some time to play with this today, looks like run.sh off of my server is having issues with /r newlines again. I'm working on it tonight, and also getting some fixes in to bugs I'm finding with sync frames (as well as putting in A380's shuffle fix).  Hoping to have a 2.18 out tonight for testing, or if not, soon.

RichN001, 2021-03-11:
glad you got it figured out.

Geektoolkit, 2021-04-02:
Let me know if we can close #59 with 2.19...it's not a 'true' live log, but it is available on the webpage now at least.

qwksilver, 2022-12-10:
these items are resolved in DynaframePro, if not resolved here specifically.

closed 2022-12-10 by qwksilver