Geektoolkit/Dynaframe3 #84: Uploads Directory

RichN001, 2021-02-06:
We should probably edit the install.sh to preserve the web/upload directory. At first I was thinking up zipping it up and moving then putting back but this would increase install or upgrade time. 

I think rather then rm Dynaframe Directory we leave it and clear it minus web/uploads

Wouldn't want to clear out peoples images they have uploaded. 

Originally I had the upload directory in Pictures, which worked fine for upload... but I could not get them to show up on the WebUI in the table... and I wanted the user to be able to see and delete them. httpServer does not like a path on the outside of the www root. 

I guess a alternative maybe could be to thumbnail them in web/uploads and put originals for display in the Pictures/Upload directory. Gives a thought for down the road that we would thumbnail the playlist images for WebUI reference then we could possibly have a real time preview in the WebUI to control.. Def a wishlist item that is. For now lets just not delete web/upload contents. ;)

Geektoolkit, 2021-02-07:
Yeah, this is a bigger problem, as you alluded to. I had some ideas of wanting to show the users a thumbnail of the current image on the frame, or to show thumbnails of a playlist, and all of that is blocked by the www access issue you hit.  
We could recursively copy the uploads folder out as part of upgrade. I guess since there is no other work done, maybe we can just do something like delete *.dll and such to clear it. This may be a better way of leaving apsettings there even.  Will have to think on it.

qwksilver, 2021-05-28:
This has been fixed for a while.

closed 2021-05-28 by qwksilver