Geektoolkit/Dynaframe3 #93: Transition is not smooth

RC-Thoughts, 2021-02-28:
Thanks for a great app!!

I have a Pi 4 4Gb with 4K monitor and 4K pics. While Dynaframe works great the transition is far away from smooth, choppy with 1-2 steps regardless of time used. (1-5 sec tested)

Am I simply asking too much with 4K display and 4K pics from the Pi4?

Original power source is used and no power- or heat-warnings are shown. Fully stock Raspbian with only Dynaframe installed from script. Tested 2.13 and 2.16, same exact thing.

Ideas?

Geektoolkit, 2021-03-01:
I highly suspect this is a combination of the library I've chosen and the limits of a Pi put together.  If I chose a more native OpenGL solution I could probably get speeds to run at 4k to do what I am doing, however it'd slow my development down so much I'd likely not release anything for a year (this is a side project of a side project).  The framework I chose gives me the ability to develop relatively quickly, but also is under development itself and is a bit heavy.  I suspect until a raspberry pi gets rev'd further to a pi5 possibly we'll be in that boat.  I notice on my pi4's that even with all of the speed increases actions such as just starting a browser can take a while, so it appears to still be fairly anemic.

I'll keep my eyes out to see if there's something I can do at some point.  On a pi3 transitions were rocky, I ended up using 20 second to 30 second transitions to smooth things out (it also creates this cool morph effect when it's that long).  

If you get a chance to try a 30 second transition, can you let me know if you see it smooth out there. Also what size screens/monitors are you using? Does 1080p (even scaling the art down) look vastly different than 4k?   I'm just curious more than anything...my only 4k screen that I've had for this has been my TV, and I'm running it at 1080p on there currently.

RC-Thoughts, 2021-03-01:
Thanks for the info. I tried 30 but since it's visibly choppy it did not look good. And yeah, on 1080p with 1080p pics all is well. No worries, I'll figure out something.

Geektoolkit, 2021-03-03:
Thanks. for now going to close this as it's technically external to Dynaframe, though I may look to see if there is anther SBC that this may run smoother on for 4k (For instance I'll have to look into an orangepi or something)

closed 2021-03-03 by Geektoolkit