Geektoolkit/Dynaframe3 #98: Rectangular behind text output

A380Coding, 2021-03-11:
Text output is white and hard to read on images with light colors e.g. pictures with snow.
Text outline or rectangular background behind text would make it easier to read.
As suggested in the other post: Adding opacity setting to the webpage and maybe color picker for the background color and opacity.
I guess the rectangle needs to be adapted to the size of the text automatically (e.g. showing file names or selecting different formats for the clock output)

RichN001, 2021-03-11:
This is a good idea, I think it might be best to just add a stroke (outline) to the text possibly with a slight shadow. Outline could have color variable.  This way it won't take away form the image to much. Also would eliminate the need to calculate the size for the box to accommodate text.

Geektoolkit, 2021-06-27:
Thanks, This will be part of the infobar/overlay rewrite.  I'm trying to decide the best way to allow for flexability of what's shown, and how it's shown. For instance if someone wants the date in the upper left, and the file name/title/author in the lower left, how to accomidate that in an easy and efficient way. I have some ideas I need to think through.